autowarefoundation/autoware #36: Runtime manager error

anhnv3991, 2015-09-14:
@kondoh2 
このようなエラーが出たのですが、何か分かりますか?
pos_downloader,pos_uploaderを起動している時だったので、それらが関係しているかもしれません。

    tc.AppendText(s)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 1850, in AppendText
    return _controls_.TextCtrl_AppendText(*args, **kwargs)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xcd in position 13: invalid continuation byte
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_core.py", line 14665, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/anh/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2650, in append_tc_limit
    tc.AppendText(s)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 1850, in AppendText
    return _controls_.TextCtrl_AppendText(*args, **kwargs)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xcd in position 13: invalid continuation byte
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_core.py", line 14665, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/anh/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2650, in append_tc_limit
    tc.AppendText(s)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 1850, in AppendText
    return _controls_.TextCtrl_AppendText(*args, **kwargs)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xcd in position 49: invalid continuation byte
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_core.py", line 14665, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/anh/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2650, in append_tc_limit
    tc.AppendText(s)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 1850, in AppendText
    return _controls_.TextCtrl_AppendText(*args, **kwargs)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xcd in position 13: invalid continuation byte
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_core.py", line 14665, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/anh/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2650, in append_tc_limit
    tc.AppendText(s)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 1850, in AppendText
    return _controls_.TextCtrl_AppendText(*args, **kwargs)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xcd in position 49: invalid continuation byte
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_core.py", line 14665, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/anh/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2650, in append_tc_limit
    tc.AppendText(s)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 1850, in AppendText
    return _controls_.TextCtrl_AppendText(*args, **kwargs)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xcd in position 13: invalid continuation byte
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_core.py", line 14665, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/anh/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2650, in append_tc_limit
    tc.AppendText(s)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 1850, in AppendText
    return _controls_.TextCtrl_AppendText(*args, **kwargs)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xcd in position 13: invalid continuation byte
[... 64 lines cut ...]
    tc.AppendText(s)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 1850, in AppendText
    return _controls_.TextCtrl_AppendText(*args, **kwargs)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xcd in position 49: invalid continuation byte
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_core.py", line 14665, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/anh/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2650, in append_tc_limit
    tc.AppendText(s)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 1850, in AppendText
    return _controls_.TextCtrl_AppendText(*args, **kwargs)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xcd in position 13: invalid continuation byte
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_core.py", line 14665, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/anh/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2650, in append_tc_limit
    tc.AppendText(s)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 1850, in AppendText
    return _controls_.TextCtrl_AppendText(*args, **kwargs)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)

kondoh2, 2015-09-14:
おそらくRuntime Managerから起動したコマンドのいづれかが、標準出力あるいは標準エラー出力へ、
0xcd という値の文字を出力した事による現象と思われます
。
Runtime Managerは起動したコマンドの出力する文字を拾ってStatusタブのGUIに表示しようとします。
そこでGUIの部品にとって不正な文字コードの判定にひっかかってるようです。
GUI部品は0x7f以下の値を期待しています。

起動コマンド側の出力箇所は特定できてませんが、Runtime Manager側で0x80以上の値を無視する対応を入れる事は可能です。

kuriking, 2015-09-14:
ごめんなさい、ヘッダのバイナリをそのまま出力している箇所があります。

shinpei0208, 2015-09-14:
修正お願いします。

kuriking, 2015-09-14:
@anhnv3991 
申し訳ございません、修正しました。お手数ですが、確認をお願いいたします。

syohex, 2015-10-05:
@anhnv3991 まだ問題が起きているでしょうか ? 問題が発生しなくなったら issueの closeをお願いします.

aalzooke, 2015-10-07:
Hi @syohex  I would like to run each node in ROS by itself, is that possible? please explain to me . 
Thank you for your help.

syohex, 2015-10-07:
> I would like to run each node in ROS by itself, is that possible?

What does `by itself` mean ?
run without runtime manager ?

aalzooke, 2015-10-07:
@syohex  can we run each node without using the runtime manager ? 
for instance, can we run dpm_ttic without using the runtime manager by typing some commands in the terminal 
i really appreciate your help.

syohex, 2015-10-08:
> can we run each node without using the runtime manager ?

```
% rosrun packagename nodename
```

For example

```
% rosrun cv_tracker dpm_ttic
```

aalzooke, 2015-10-09:
@syohex Thank you for being so helpful, Also i would like to know which file that storage the source code for calculating the distance for car and Pedestrian in autonomous vehicle?

[image]

syohex, 2015-10-10:
> Also i would like to know which file that storage the source code for calculating the distance for car and Pedestrian in autonomous vehicle as showing in the picture below ?

CC: @manato, @pdsljp

amc-nu, 2015-10-10:
Yosh-san

Please look at the CV_tracker specially range fusion and its library, under computing, perception.

Greetings

Syohei YOSHIDA さんは書きました :

closed 2015-11-12 by shinpei0208