autowarefoundation/autoware #62: compile error

TomohitoAndo, 2015-09-25:
@syohex

最新版のコンパイル時に以下のようなエラーが出ましたが、何のエラーかわかりますか。
```
/home/ando/Autoware/ros/src/computing/perception/detection/packages/cv_tracker/nodes/dpm_ocv/dpm_ocv.cpp: In constructor ‘objectDetect::objectDetect()’:
/home/ando/Autoware/ros/src/computing/perception/detection/packages/cv_tracker/nodes/dpm_ocv/dpm_ocv.cpp:76:39: error: ‘use_gpu’ was not declared in this scope
  if (!private_nh_.getParam("use_gpu", use_gpu)) {
                                       ^
[ 93%] Built target points2vscan
Scanning dependencies of target kf_track
[ 93%] Building CXX object computing/perception/detection/packages/cv_tracker/CMakeFiles/kf_track.dir/nodes/kf_track/kf_track.cpp.o
make[2]: *** [computing/perception/detection/packages/cv_tracker/CMakeFiles/dpm_ocv.dir/nodes/dpm_ocv/dpm_ocv.cpp.o] エラー 1
make[1]: *** [computing/perception/detection/packages/cv_tracker/CMakeFiles/dpm_ocv.dir/all] エラー 2
make[1]: *** 未完了のジョブを待っています....
```

syohex, 2015-09-25:
c5afe0a138b5991572dcc2a4e1b043f9857b59ff の修正が不十分です(`ifdef`で囲むべきところが囲まれていません).

下記のパッチを適用して再度試してみてください.

```patch
diff --git a/ros/src/computing/perception/detection/packages/cv_tracker/nodes/dpm_ocv/dpm_ocv.cpp b/ros/src/computing/perception/detection/packages/cv_tracker/nodes/dpm_ocv/dpm_ocv.cpp
index 8a027be..364ec8b 100755
--- a/ros/src/computing/perception/detection/packages/cv_tracker/nodes/dpm_ocv/dpm_ocv.cpp
+++ b/ros/src/computing/perception/detection/packages/cv_tracker/nodes/dpm_ocv/dpm_ocv.cpp
@@ -73,9 +73,11 @@ objectDetect::objectDetect() :
 		object_class = "car";
 	}
 
+#if defined(HAS_GPU)
 	if (!private_nh_.getParam("use_gpu", use_gpu)) {
 		use_gpu = false;
 	}
+#endif
 
 	std::string default_model;
 	// switch (type) {

```

TomohitoAndo, 2015-09-25:
素早い対応ありがとうございます。
上記の修正を加えてコンパイルしてみたところ、通ることが確認できました。
この修正、どうすればいいでしょうか。

syohex, 2015-09-25:
#64に修正を出しました. ご確認ください.

manato, 2015-09-25:
お手数おかけいたしました。
ありがとうございます。

closed 2015-09-25 by syohex