autowarefoundation/autoware #89: rosのmake時のエラーについて

lauguste, 2015-10-17:
Autoware/ros/でのmake時にエラーが出るようです。
環境: Ubuntu14.04(実機上、VM上)
リビジョン: 投稿時点で最新のリビジョン(53b10e)
現象: rosの./catkin_make_release時に下記ログのエラーが出ます。
インストール手順: README.mdの手順通りに行っています。
(OptionalのCUDA, FlyCapture2のインストールは行っていません。)

-----------------------------catkin_make時の標準出力(エラー部分抜粋)-----------------------------
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by
  "velodyne_pointcloud" with any of the following names:

    velodyne_pointcloudConfig.cmake
    velodyne_pointcloud-config.cmake

  Add the installation prefix of "velodyne_pointcloud" to CMAKE_PREFIX_PATH
  or set "velodyne_pointcloud_DIR" to a directory containing one of the above
  files.  If "velodyne_pointcloud" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  computing/perception/localization/packages/ndt_localizer/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

---------------------------------------------CmakeError.log---------------------------------------------
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec2837461572/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec2837461572.dir/build.make CMakeFiles/cmTryCompileExec2837461572.dir/build
make[1]: Entering directory `/home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec2837461572.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec2837461572.dir/CheckSymbolExists.c.o   -c /home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTryCompileExec2837461572
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec2837461572.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec2837461572.dir/CheckSymbolExists.c.o  -o cmTryCompileExec2837461572 -rdynamic
CMakeFiles/cmTryCompileExec2837461572.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[1]: Leaving directory `/home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec2837461572] Error 1
make: *** [cmTryCompileExec2837461572/fast] Error 2

File /home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec2347961959/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec2347961959.dir/build.make CMakeFiles/cmTryCompileExec2347961959.dir/build
make[1]: Entering directory `/home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec2347961959.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTryCompileExec2347961959.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-2.8/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec2347961959
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec2347961959.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTryCompileExec2347961959.dir/CheckFunctionExists.c.o  -o cmTryCompileExec2347961959 -rdynamic -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: Leaving directory `/home/lauguste/Workspace/Autoware/ros/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec2347961959] Error 1
make: *** [cmTryCompileExec2347961959/fast] Error 2

syohex, 2015-10-17:
`ros-indigo-velodyne-pointcloud`パッケージはインストールされているでしょうか ?
インストールしていなければ, インストールした後再度試していただけないでしょうか ?

lauguste, 2015-10-17:
早速のご回答ありがとうございます。
ご回答頂いたパッケージをインストール後、makeが進みましたが、下記ログの別のエラーが発生します。
なお、OpenCVのバージョンは3.0.0、g++のバージョンは4.8.4です。
度々申し訳ありませんが、ご助力頂ければ幸いです。

-----------------------------catkin_make時の標準出力(エラー部分抜粋)-----------------------------
[  4%] Building CXX object computing/perception/detection/lib/image/dpm_ocv/CMakeFiles/libdpm_ocv.dir/cpu/dpm_ocv_cpu.cpp.o
In file included from /home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/cpu/dpm_ocv_cpu.cpp:3:0:
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:7:65: error: expected class-name before ‘{’ token
 class DPMOCVCPULatentSvmDetector : public cv::LatentSvmDetector {
                                                                 ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:9:17: error: ‘CvLatentSvmDetector’ was not declared in this scope
     std::vector<CvLatentSvmDetector*> detectors; // this is private member in cv::LatentSvmDetector
                 ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:9:37: error: template argument 1 is invalid
     std::vector<CvLatentSvmDetector*> detectors; // this is private member in cv::LatentSvmDetector
                                     ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:9:37: error: template argument 2 is invalid
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:13:52: error: ‘ObjectDetection’ was not declared in this scope
     void detect( const cv::Mat& image, std::vector<ObjectDetection>& objectDetections,
                                                    ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:13:67: error: template argument 1 is invalid
     void detect( const cv::Mat& image, std::vector<ObjectDetection>& objectDetections,
                                                                   ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:13:67: error: template argument 2 is invalid
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:20:65: error: expected class-name before ‘{’ token
 class DPMOCVGPULatentSvmDetector : public cv::LatentSvmDetector {
                                                                 ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:23:17: error: ‘CvLatentSvmDetector’ was not declared in this scope
     std::vector<CvLatentSvmDetector*> detectors;
                 ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:23:37: error: template argument 1 is invalid
     std::vector<CvLatentSvmDetector*> detectors;
                                     ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:23:37: error: template argument 2 is invalid
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:33:21: error: ‘ObjectDetection’ was not declared in this scope
         std::vector<ObjectDetection>& objectDetections, float overlapThreshold,
                     ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:33:36: error: template argument 1 is invalid
         std::vector<ObjectDetection>& objectDetections, float overlapThreshold,
                                    ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/include/dpm_ocv/dpm_ocv.hpp:33:36: error: template argument 2 is invalid
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/cpu/dpm_ocv_cpu.cpp:32:35: error: ‘CvLSVMFilterObject’ does not name a type
 extern int getMaxFilterDims(const CvLSVMFilterObject **filters, int kComponents,
                                   ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/cpu/dpm_ocv_cpu.cpp:32:56: error: ISO C++ forbids declaration of ‘filters’ with no type [-fpermissive]
 extern int getMaxFilterDims(const CvLSVMFilterObject **filters, int kComponents,
                                                        ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/cpu/dpm_ocv_cpu.cpp:36:54: error: ‘CvLSVMFilterObject’ does not name a type
                                                const CvLSVMFilterObject **filters,
                                                      ^
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/cpu/dpm_ocv_cpu.cpp:36:75: error: ISO C++ forbids declaration of ‘filters’ with no type [-fpermissive]
                                                const CvLSVMFilterObject **filters,
                                                                           ^
[... cut ...]
/home/lauguste/Workspace/Autoware/ros/src/computing/perception/detection/lib/image/dpm_ocv/cpu/dpm_ocv_cpu.cpp:527:15: warning: ‘CvSeq* cvLatentSvmDetectObjects(IplImage*, int*, CvMemStorage*, float, int, double, int, int, int)’ defined but not used [-Wunused-function]
 static CvSeq* cvLatentSvmDetectObjects(IplImage* image,
               ^
make[2]: *** [computing/perception/detection/lib/image/dpm_ocv/CMakeFiles/libdpm_ocv.dir/cpu/dpm_ocv_cpu.cpp.o] Error 1
make[1]: *** [computing/perception/detection/lib/image/dpm_ocv/CMakeFiles/libdpm_ocv.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

syohex, 2015-10-17:
OpenCV3には現在対応していません. OpenCV 2.X(2.4.8等)を利用してください.

lauguste, 2015-10-20:
ご回答ありがとうございます。
本件READMEにも記載されていましたが、見落としていました。

本件解決したため、クローズとさせていただきま

closed 2015-10-20 by lauguste