autowarefoundation/autoware #92: rosのmake時のエラーについて(2)

lauguste, 2015-10-20:
度々失礼します。別環境にてAutowareのインスートルを行おうとしたところ、Autoware/ros/のmake時にエラーが出ました(前回質問時のものと異なるものです)。度々申し訳ありませんが、ご助言いただけると大変助かります。

環境: Ubuntu14.04(実機上)
リビジョン: 投稿時点で最新のリビジョン(53b10e)
現象: rosの./catkin_make_release時に下記ログのエラーが出ます。
インストール手順: README.mdの手順通りに行っています。
(OptionalのCUDA, FlyCapture2のインストールは行っていません。)

*本件前回質問時の環境では起きていませんでした。前回との環境の違いはVM上であること、proxy配下にあることです。

複数台のPCで試してみましたが、いずれも同様のエラーが発生しています。
以下、標準エラー出力の抜粋です。

Cmake Error at /home/laugust-dev/Autoware/ros/devel/share/glviewer/cmake/glviewerConfig:141 (message):
  Project 'calibration_camera_lidar' tried to find library 'glviewer'. The
  library is neither a target nor built/installed propery. Did you compile project 'glviewer'? Did you find_package() it before the subdirectory
  containing its code is included?

syohex, 2015-10-20:
ビルドログを gist等に全部貼り付けてもらえないでしょうか ?
(量が多くなければここでも構いませんですが. あとログ等をはりつけるときは, バッククォート 3つで囲っていただけると助かります.).

アーキテクチャはなんでしょうか ? x86_64でしょうか ?
i386(32bit)だとそのようなエラーが出て, 失敗しそうに見えます.

syohex, 2015-10-20:
32bit環境だと再現しました.
パッチは下記に作成しました.

https://gist.github.com/syohex/dbbe8b11f8340125e678

32bit環境対応が必要であるならば本体にマージしますので, お知らせください.
(Qt5を Qt4で全部書き換えることができればこのような対応はそもそも必要ないのですが...)

syohex, 2015-10-22:
@lauguste 最新版だと 32bitでもビルドできるようになっているので, 最新版に更新してから再度確認していただけないでしょうか ?

lauguste, 2015-10-22:
ご対応大変感謝します。本日夜再度試してみて追ってご連絡いたします。

lauguste, 2015-10-24:
Ubuntu 14.04.03 32bitで試しました。
Qt周りでエラーが起きているように見えます。
Qtのバージョンは5.6と、5.2.1で試しており、下記は5.2.1の時のエラーです。
https://gist.github.com/lauguste/dbe0ff8ac21d8d1861f3

syohex, 2015-10-24:
Qt5は Ubuntuパッケージのものを利用しているでしょうか(`qtbase5-dev`, `libqt5opengl5-dev`) ?
そうでなければ, パッケージを aptコマンドでインストールしてそれを利用していただけないでしょうか.

syohex, 2015-10-24:
新しいソースコードでは Qt5のコンパイルフラグ(include path等), リンクフラグ(リンクするライブラリ名等)は pkg-configを使って取得しています(現状 ROSが Qt5をサポートしていないためこのような面倒なことをしている), そのため pkg-configの設定ファイル(拡張子 `.pc`)が pkg-configコマンドが見つけられる位置になくてはなりません. エラーは pkg-configの設定が見つけられていないものと思われます.

syohex, 2015-10-24:
あと 32bit環境での動作実績は皆無だと思いますので, 用意できるのであれば 64bit環境の方がよいと思います.

lauguste, 2015-10-24:
Qt５は以前の手順書にあった方法でインストールしていました。
```
sudo apt-get build-dep -y qt5-default
sudo apt-get install -y build-essential perl python git
sudo apt-get install -y  "^libxcb.*" libx11-xcb-dev libglu1-mesa-dev libxrender-dev libxi-dev
sudo apt-get install -y  flex bison gperf libicu-dev libxslt-dev ruby
sudo apt-get install -y libssl-dev libxcursor-dev libxcomposite-dev libxdamage-dev libxrandr-dev libfontconfig1-dev
sudo apt-get install -y libasound2-dev libgstreamer0.10-dev libgstreamer-plugins-base0.10-dev

cd ~/Workspace
git clone git://code.qt.io/qt/qt5.git
```
aptで入れて試してみます。
なお最初にご指摘頂いた際に64bit環境に切り替えており、上記は動作検証として行っています。

syohex, 2015-10-24:
aptで入れた場合でも同様のエラーが出るのでしょうか ?

syohex, 2015-11-13:
現在以下の環境でビルドできることは確認できています.
- Ubuntu 14.04 64bit ROS Indigo
- Ubuntu 15.04 64bit ROS Jade
- Ubuntu 14.04 32bit ROS Indigo
- Ubuntu 15.04 32bit ROS Jade

https://github.com/CPFL/Autoware/blob/master/README.md

パッケージはトップページの READMEに従って行ってください.

問題が解決しない, READMEに不備がある場合などは本 issueを reopenするなどして
知らせてもらえればよいかと思います.

closed 2015-11-13 by syohex

Soroosh129, 2017-06-02:
I have the same problem now when installing on Ubuntu 16.04. Can you please help me with that?