autowarefoundation/autoware #111: cuda_memcheck reports error in calc_a_score

KenYN, 2015-11-06:
Running `cuda_memcheck`, I found that [in `dpm_ttic`'s `GPU_function.cu`](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/detection/lib/image/dpm_ttic/gpu/GPU_function.cu#L891) there is an out-of-bounds GPU memory read.

When `component_jj` equals `NoC-1`, `int ssize1 = *((int *)pointer_ssize + sizeof(int));` - or the clearer to me `int ssize1 = ssize_array[component_jj+1];` - accesses one past the end of the array, that is `ssize_array[NoC]`, as [the allocation in `get_boxes.cpp`](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/detection/lib/image/dpm_ttic/gpu/get_boxes.cpp#L144) is only for `NoC` elements.

I also note similar issues in CPU code with the `rm_size_array` [here](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/detection/lib/image/dpm_ttic/gpu/get_boxes.cpp#L663) and [here](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/detection/lib/image/dpm_ttic/gpu/get_boxes.cpp#L967).

* Furthermore, shouldn't `(int *)pointer_ssize + sizeof(int)` be `(int *)pointer_ssize + 1` or `(int *)(pointer_ssize + sizeof(int))`?

syohex, 2015-11-06:
Could you send pull-request ?

shinpei0208, 2015-11-12:
What's the status of this issue?

syohex, 2015-11-13:
This is related to #112

KenYN, 2015-11-13:
Thinking about this problem, I believe the problem is as linked in [get_boxes.cpp line 144](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/detection/lib/image/dpm_ttic/gpu/get_boxes.cpp#L144)

    res = cuMemAlloc(&ssize_dev, (NoC+1)*sizeof(int));

And at [line 396](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/detection/lib/image/dpm_ttic/gpu/get_boxes.cpp#L396):

    int **rm_size_array = (int **)malloc(sizeof(int *)*(L_MAX+1));

Both seem to need a `+1` on the upper bound. However, I don't have time to test this out, so I cannot send a pull-request.

BTW, in [computing.yaml, line 43](https://github.com/CPFL/Autoware/blob/master/ros/src/util/packages/runtime_manager/scripts/computing.yaml#L43), if you add 

    cmd  : cuda-memcheck roslaunch cv_tracker dpm_ocv.launch

You can easily run the CUDA memory leak checker.

closed 2020-05-29 by gbiggs