autowarefoundation/autoware #173: autoware directory

bashrc, 2015-12-18:
I notice that many files contain a reference to an ~/.autoware directory. Should this not be ~/Autoware ?

RicWg, 2016-05-20:
they are different.
just create ~/.autoware, download my_launch.sh and run it under it, unzip the bag files , it will has ./data inside
~/Autoware has all the code

DamonMIN, 2016-05-31:
@RicWg 
In Quick Start, there are *.launch files in map/Locatilation/Mission planning/Motion planning. Where are these launch files from? Can you explain to me how the autoware software works? Thank you.

RicWg, 2016-05-31:
there are some sample data 3D( map of Moriyama in Nagoya/ ROSBAG data of Moriyama driving) here https://github.com/CPFL/Autoware, unzip them to .autoware.

the doc provides some basic instruction. try to use google translate to read them.

RicWg, 2016-05-31:
BTW, you can join this group in the slack: deepdriving.slack.com, we have couple people working on this topic.

DamonMIN, 2016-06-01:
@RicWg 
To start using Slack, i'll need to be invited by theTeam Administrator.
So,can you send me a link to invite me ?

RicWg, 2016-06-02:
@DamonMIN 
I can send a link to your email, what's it?

DamonMIN, 2016-06-03:
@RicWg 
When I tried to join the group in the slack: deepdriving.slack.com using my personal email 2577876871@qq, slack sent me an email.
/////////////////////////////////////////////////////////////////////////
We searched for Slack teams you've already joined or are allowed to join, using 2577876871@qq.​com.

Unfortunately, we couldn't find any teams for that address.

If you're looking for a specific team, you can try contacting your Team Administrator for an invitation. You can also try a different email, or forge ahead and create a new team!

Please let us know if you have any other questions or feedback. And thanks for using Slack!
////////////////////////////////////////////////////////
Which email address should I use? Or  can team administrator send me an invitation?

RicWg, 2016-06-04:
invitation sent

yukikitsukawa, 2016-06-08:
@RicWg I am interested in your group. Can you invite me to the Slack team?

vlnguyen92, 2016-07-21:
@RicWg: Can you invite me to the slack team as well?

vlnguyen92, 2016-07-21:
My email is nguyenvanlinh1992@gmail.com

travelbureau, 2016-07-21:
@RicWg: Could I have an invite... mokelly@seas.upenn.edu

droter, 2016-10-13:
Same issue.

I am following the manual and got to 2.1 there is no "run" or ~Autoware directory.  The instructions just say to make .autoware.  Could I get on the slack channel also?

Thanks,

Matt

mslavescu, 2017-01-06:
@RicWg what is the Slack team for Autoware? Could you please send me an invite at marius at gtarobotics.com.

I'm running the Open Source Self Driving Car (OSSDC) Initiative and need help integrating Autoware there.

To join OSSDC Slack team and OSSDC Github check this page:
http://ossdc.org/

closed 2017-05-29 by yk-fujii

PeterZheFu, 2017-07-24:
@RicWg  May I know if I can be added to the slack channel, too? My email is peterzhefu@ece.ucsb.edu, or preferrably at peterzhefu@gmail.com    Thank you.