autowarefoundation/autoware #190: Is the sample moriyama data sufficient to reproduce the video demos?

bashrc, 2016-01-05:
So far this looks like maybe a smaller section of a bigger data set.

manato, 2016-01-06:
@bashrc 
As you mentioned, sample moriyama data doesn't include images.
But we didn't modify any other data like  LIDAR and GNSS log.
Could you specify which video you intend to re-enact?
(Because I want to make sure that video you want to re-enact is made from sample data we release.)

bashrc, 2016-01-06:
As an initial try I was going to attempt to reproduce the first video. https://www.youtube.com/watch?v=ztUtN3ZG6N8

manato, 2016-01-06:
@bashrc ,
Thank you for providing the link.
I'm sorry but that video doesn't seem to be made from sample data you can download.
But the data you can download is sufficient to reproduce the video excluding image processing because it contains LIDAR data that is all you need to reproduce the video demo.
Of course the result will not be exactly same as video but same function excluding image processing in video should work with it.

bashrc, 2016-01-06:
It looks like there's more data than I thought. I had assumed that once the bag replay had reached 100% then that was the end, but it looks like it isn't.

shinpei0208, 2016-01-09:
@shohex, @kuriking,

Isn't it a bug? ROSBUG should end when it reaches on Runtime Manager.

shinpei0208, 2016-01-09:
@syohex, @kuriking,

Isn't it a bug? ROSBUG should end when it reaches on Runtime Manager.

bashrc, 2016-01-09:
Perhaps.

kuriking, 2016-01-12:
Please forgive me for writing in Japanese.

$ rosparam set use_sim_time true
$ rosbag play sample_moriyama_150324.bag --clock
と手動で動かした場合も、Runtime Managerから起動した場合も、479秒経過すると、
RvizのROS TimeとROS Elapsedは停止します。(rosbag playは終了します。)
Runtime ManagerのPlayボタンが押せない状態のままになっている、というのはあります。

syohex, 2016-01-12:
Sorry I don't understand ROSBUG well.

shinpei0208, 2016-01-13:
@bashrc 

According to our checking, when the rosbag reaches 100%, the simulation ends. What was the problem again?

shinpei

beetleskin, 2016-04-25:
Well .. I'm trying to reproduce this tutorial with the data you provided, but I can't even find the spot you marked where you set the initial pose in the video. Further, there is no tf data for the /velodyne frame, therefore the data can not be used.

mihirmody77, 2016-08-12:
I am trying to "Detection"  using Autoware.

How to get captured raw images for "Detection" that is used for "image processing e.g. Car detection"?

Is it part of sample_moriyama_150324.bag ?

If this is not part of sample_moriyama_150324.bag , can you upload sample raw images from camera?

Regards,
Mihir

kuriking, 2016-08-14:
Raw images are not included in sample_moriyama_150324.bag because of privacy.
You cant get raw images by UVC camera, Grasshopper 3 or Ladybug 5.

wgcv, 2016-10-13:
Where i can get all file to start?

closed 2017-05-29 by yk-fujii