autowarefoundation/autoware #195: Sample data links are broken

rg4444, 2016-01-13:
We are really enjoying Autoware software and considering to use it in our project.
However we are trying to test the installation of software with real data, but there is problem that  links do not work:
3D map of Moriyama in Nagoya http://db3.ertl.jp/autoware/sample_data/sample_moriyama_data.tar.gz 
ROSBAG data of Moriyama driving http://db3.ertl.jp/autoware/sample_data/sample_moriyama_150324.tar.gz 
Would be great if we could get those sample files elsewhere.
Thank you in advance!

shinpei0208, 2016-01-14:
They should work now. Thanks.

closed 2016-01-14 by shinpei0208

rg4444, 2016-01-14:
Unfortunately not. Still got connection timeout message. We are located in Europe. Maybe some regional restriction from hosting provider?

yukikitsukawa, 2016-01-14:
@rg4444 Can you try again? I modified the restriction policy.

syohex, 2016-01-14:
I confirm that I can access above links(I could not download before some minutes )

rg4444, 2016-01-14:
Yes, now it is working perfectly! Thank you!