autowarefoundation/autoware #221: Correct building process

syohex, 2016-01-29:
- [ ] Fix cycle dependency issues
- [ ] Declare package dependency correctly. Some dependencies are not declared for avoiding build errors now.
- [ ] Remove build warnings. There are some build warnings about rostest and roslaunch.

CC: @manato @yukikitsukawa @syouji

syohex, 2016-02-01:
There are following cycle dependencies(I may miss some cycle dependencies :-( )

- cv_tracker <-> fusion
- cv_tracker <-> kf
- cv_tracker <-> points2image
- cv_tracker <-> scan2image
- ndt_localizer <-> runtime_manager
- tablet_socket <-> vehicle_socket

syohex, 2016-02-02:
Autoware package dependencies are very, very complex. Especially library design is terrible. It seems it is impossible without rewriting many parts of code :disappointed:.

syohex, 2016-02-02:
Most of messages have unique names but LaneArray.msg is duplicated.

- [ros/src/computing/planning/motion/packages/waypoint_follower/msg/LaneArray.msg](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/planning/motion/packages/waypoint_follower/msg/LaneArray.msg)
- [ros/src/data/packages/map_file/msg/LaneArray.msg](https://github.com/CPFL/Autoware/blob/master/ros/src/data/packages/map_file/msg/LaneArray.msg)

We need to rename if we manage all messages into one package.

And there are some same name messages except case.
- `dtlane.msg` and `DTLane.msg`
- `lane.msg` and `Lane.msg`
- `waypoint.msg` and `Waypoint.msg`

It is confused if they are in one directory.

shinpei0208, 2016-02-02:
Let's rewrite the code in Autoware version 2. We plan to redesign GUI as
well. Usuda-san, please list this up in TODO work.

syohex, 2016-02-11:
Sorry I write Japanese at first.

### パッケージ間の依存を増やさないための注意

#### ライブラリのヘッダファイルで messageから生成されるヘッダファイルを includeしない

関連する関数を使わないパッケージでも, それを includeしてしまうと, その messageを生成するパッケージに依存してしまうことになる.

#### ライブラリで messageから生成されるヘッダファイルを includeしない

それをリンクするパッケージが間接的に messageを生成するパッケージに依存してしまう. メッセージで渡ってきたデータの処理をライブラリに切り出したい場合は, メッセージの中のデータを渡すようにするなどする(std::vector型等). もしくは別途定義したメッセージに依存しない class, struct移し替える等する

#### むやみにライブラリに切り出さない

汎用的なもの以外は切り出さない方がいい. パッケージ間の依存を返って高めてしまう可能性がある. 別パッケージにするほど有用かをまず考える.

#### その他気になったところ

##### ライブラリにグローバル変数がある
- ライブラリが状態を持つ
 - 複数スレッドで使えるのか
 - 状態を引数として渡すようにできるかを検討する

##### ライブラリ関数の引数がない, 戻り値が void
-  何のために呼び出している ?
- ライブラリの状態を変化させている ?
- ライブラリ関数としては引数だけに依存するものが望ましい.

#### 名前が適当すぎる
- namespaceを使う or ライブラリの prefixをつける

closed 2018-04-05 by yk-fujii