autowarefoundation/autoware #240: How to show sample velodyne data on RViz?

odakoh, 2016-02-15:
When I tried to playback sample rosbag data on the latest Autoware (a982bf0543d4cbce1c37cccc09a98b79cb5a4f0c) and show the data on RViz, I got error _"For frame [velodyne]: Frame [velodyne] does not exist"_ in "Points Raw -> Status: Error -> Transform[sender=unknown_publisher]".

Is there anything I have to do before playback?
My procedure steps were like the below
- load map/vector/tf
- load rosbag
- turn on nmea2tfpose/ndt_matching
- play

Besides, on the latest Autoware GUI (on my environment), there is no option "Clock" and "Sim Time" in Simulation tab. How can I set those parameters?

manato, 2016-02-15:
@odakoh ,
Thank you for contact.
On the latest Autoware, additional steps are required so that NDT_matching works correctly.
Please try steps below:
* load map/vector/tf
* load rosbag
* **Go "setup" tab and push 2 buttons there ("TF" and "Vehicle model")**
* turn on nmea2tfpose/ndt_matching
* play

"Clock" and "Sim Time" are integrated into "Play" button because those are turned on every time we use rosbag.  Those are turned on automatically when you push "Play" button.

odakoh, 2016-02-15:
@manato 
Thank you for your quick reply.
With your proposed new step, I got the different error message like the below.

> For frame [velodyne]: No transform to fixed frame [world].  TF error: [Lookup would require extrapolation into the past.  Requested time 1427157658.2563    22048 but the earliest data is at time 1455564217.206791698, when looking up     transform from frame [velodyne] to frame [world]]
]

There is no Vehicle Model file in sample data, so is it acceptable to keep empty in Vehicle Model input area and just push the button?

manato, 2016-02-15:
I think the error may be caused by how time is treated in ROS.
As "Sim Time" option will not be enable unless Play button is pushed, please try these step instead of "load rosbag":
* fill the input area for rosbag by choosing bag data from "Ref"
* **Push "Play" button, then push "Pause" immediately** 

Subsequent process can recognize that they should use time recorded in rosbag (not current one) by these step.

> There is no Vehicle Model file in sample data, ...

Vehicle Model is included Autoware itself. Please specify `[Autoware_installed_path]/Autoware/ros/src/.config/model/prius.dae` by default.

odakoh, 2016-02-16:
Hello @manato,

Thank you for your response.
My issue was resolved with your step "push pause immediately playback.
After changed my steps like the below, everything was fine.

1. Ref rosbug file, and push play and pause it quickly.
2. Load map/vector/tf file
3. Enable TF/Model in setup tab
4. Run RViz
5. Resume rosbug playback

closed 2016-02-16 by odakoh

Tikitikitiki, 2018-08-08:
Hi @manato, I got the same tf error 

For frame [velodyne]: No transform to fixed frame [world].  TF error: [Lookup would require extrapolation into the past.  Requested time 1427157730.723200000 but the earliest data is at time 1533730917.031480030, when looking up transform from frame [velodyne] to frame [world]]

But it does not help when press play then pause immediately in the beginning.