autowarefoundation/autoware #262: Processor load info missing

rg4444, 2016-03-27:
We running Autoware on several different Ubuntu 14.04, ROS Indigo hosts, and none of them display real-time processor load, as seen on demo videos.
Installation done as described in instruction. 
Whet we are missing, or this functionality is not included in GitHub source?

[image]
[image]

syohex, 2016-03-30:
> Whet we are missing, or this functionality is not included in GitHub source?

Yes it is included in Autoware source tree by default. Did you get any warnings or error messages ?

1. Launch Runtime Manager by `rosrun runtime_manager runtime_manager_dialog.py` from terminal emulator
2. Can you see any warnings or error in terminal emulator ?

rg4444, 2016-03-30:
[rospack] Error: package 'runtime_manager' not found

syohex, 2016-03-30:
How about following steps ? (If you use zsh, please load zsh scripts instead of bash scripts)

1. `source ./devel/setup.bash`
2. `cd ~/Autoware/ros`
3. `source ./devel/setup.bash`
4. `rosrun runtime_manager runtime_manager_dialog.py`

rg4444, 2016-03-30:
Yes,, now its work
Output is: (and GUI window pop-up without processor info)

loading param.yaml
loading qs.yaml
loading setup.yaml
loading map.yaml
loading sensing.yaml
loading computing.yaml
loading interface.yaml
loading data.yaml
loading simulation.yaml
loading status.yaml
creating item node list .cmd_param file is required
cmd_param file is required
cmd_param file is required
cmd_param file is required
cmd_param file is required
...cmd_param CalibrationFile is required
cmd_param file is required
[camera.launch] is neither a launch file in package [pointgrey_camera_driver] nor is [pointgrey_camera_driver] a launch file name
The traceback for the exception was written to the log file
cmd_param CalibrationFile is required
cmd_param calibration is required
cmd_param calibration is required
cmd_param calibration is required
cmd_param calibration is required
.....
loading topics.yaml
Exception in thread Thread-7:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/user/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 1310, in top_cmd_th
    fv = float(v)
ValueError: invalid literal for float(): 0,0

syohex, 2016-03-30:
Thanks for information.

syohex, 2016-04-01:
@rg4444 Could you show us output of `env COLUMNS=512 top -b -n 2 -d 0.1` ?

rg4444, 2016-04-01:
See attached output file:

[output.txt](https://github.com/CPFL/Autoware/files/199091/output.txt)

syohex, 2016-04-01:
Thanks. This is locale issue. We only assume number expression in English(123.45), not Spanish(123,45).

syohex, 2016-04-01:
Sorry I use Japanese in this comment.

@kondoh2 `top`コマンドの解析箇所が localeにより例外が発生してしまっています. スペイン方式(`de_DE.UTF-8`等)だと小数点数の表示形式が `123,45`(小数点がカンマ)となり, これを Pythonの `float`関数に与えると例外が発生します(Pythonの `float`関数は英米方式 `123.45`しか受け付けない). 解決方法としては下記の URLのようにカンマをピリオドで置き換える, `float`でなく `locale`パッケージの `atof`関数を使う, `top`コマンドを `LC_ALL=C`等で実行する方法が考えられます. `float`が使われている箇所が複数存在し, 私では理解が及ばない部分もありますので, 修正案を検討いただければと思います. よろしくお願いします.

http://stackoverflow.com/questions/6633523/how-can-i-convert-a-string-with-dot-and-comma-into-a-float-number-in-python

rg4444, 2016-04-01:
We change Locale on Ubuntu 14.04 settings to English (United States) and now processor load and list of processes are visible. Thank you!

closed 2016-04-01 by rg4444

kondoh2, 2016-04-01:
ご指摘ありがとうございます。
了解です。
検討します。

syohex, 2016-04-04:
This issue was fixed by #268 change.