autowarefoundation/autoware #263: Custom Vector map creation

rg4444, 2016-03-27:
From tutorials and demo data it is possible to replicate point cloud (from sequence of .pcd files) and Vector map (from set of .csv files - crosswalk.csv, stopline.csv, zebrazone.csv etc).
In Rviz vector map .csv file content appears as Namespaces in Vector map marker topic.
What editing tool, GUI should be used to add or change objects in vector maps? 
Or what are the steps to create new set of objects- like "Stop lines" to add to vector map. 
What are the tools (opensource?) used to create vector maps of sample data?

sgzzgit, 2016-05-18:
Have you got the answer to this question already?

nsakar, 2016-05-18:
Can you provide some instructions on how to create a vector map with landmarks and lane marks?

I have been able to generate a path via waypoint saver but not a vector map of landmarks.

Regards

Nandini

shinpei0208, 2016-05-19:
We don't have a function to create a vector map. We ask a map company to
construct it for us. Having said that, there is some work in progress to
create a vector map by Autoware itself, though the timeline of milestones
is not clear yet.

rg4444, 2016-05-19:
Thank you for answer.
It is very valuable function - to generate / edit / create vector maps.
We are now researching/developing such a "Autoware Server" able to perform vector map semiautomatic creation.
Would be happy to join efforts with other developers

RicWg, 2016-05-20:
any try to work with map vendors like civilmap, openstreetmap, etc?

maths4513, 2017-11-07:
@shinpei0208
Is vector map files .csv is a stantard gis data format ? Is there any reference documents about this "vector map". hoping for your reply.  :+1:

dejanpan, 2017-11-10:
@RicWg working with other vendors is not happening yet but everybody is always welcome. Autoware is open source and thus for evveryone.

@maths4513 this is a custom format, provided by a company http://www.aisantec.co.jp/english/ (http://www.aisantec.co.jp/english/mms.html).
All the shapes in the .csv files are geo-referenced (in ECEF I think).

There is also an online tool where you can create a basic vector map yourself: https://maptools.tier4.jp/.

closed 2017-11-10 by dejanpan