autowarefoundation/autoware #269: Autoware on 32bit or ARM systems

rg4444, 2016-04-04:
We are trying to launch Autoware on Nvidia Jetson TK1, which is 32bit and Arm architecture.
Most of drivers and libs are installed pretty good, however we stuck on "glviewer" missing while ./catkin_make_release

We found out this gist: "Autoware patch for building on Ubuntu 14.04 32bit " https://gist.github.com/syohex/dbbe8b11f8340125e678 , but were unable to patch latest build files. Reason might be,that patch is 6 month old, but cmake files are newer.
Any workarounds on that?

syohex, 2016-04-04:
> Most of drivers and libs are installed pretty good, however we stuck on "glviewer" missing while ./catkin_make_release

glviewer is built on only x86 architecture now(Its condition branch is [here](https://github.com/CPFL/Autoware/blob/3ac0843ee1d39de0dab8da2dfc197e9f53f6f254/ros/src/util/packages/RobotSDK/glviewer/CMakeLists.txt#L40)). I tried to build it on Jetson TK1 one and half years ago but I could not built it.

rg4444, 2016-04-04:
It could be good if somebody could find any workaround on that. 
All Nvidia automotive prototyping hardware like Jetson TK1, TX1 and Drive PX are based on ARM. Could be great hardware to test and run Autoware.

winnerlin, 2016-12-02:
After look into the build process, there I found the issue in CMakeLists.txt
within Autoware/ros/src/util/packages/RobotSDK/glviewer/CMakeLists.txt
if("${ARCHITECTURE}" STREQUAL "x86_64" OR "${ARCHITECTURE}" STREQUAL "i686")
include_directories(
...

change into
if("${ARCHITECTURE}" STREQUAL "x86_64" OR "${ARCHITECTURE}" STREQUAL "i686" OR "${ARCHITECTURE}" MATCHES "^arm")

But I still encounter some other issue.

SakataGintoki435, 2017-08-08:
Is there any resolution to this? I am stuck on the same thing, with an ARM board.

CodeIsWorld, 2017-08-31:
when i try to run this project in nvidia jetson TX2,I also meet this question.how to solve this problem?

CodeIsWorld, 2017-09-06:
I send the problem to autoware@googlegroups.com,and I got the answer" "
`It is necessary to change CMakeFiles.txt of glviewer package.
Please refer to the URL below:
[https://github.com/CPFL/Autoware/blob/a0bbcad24f6b648380f63fbe8e9d76b9b2dc0798/ros/src/util/packages/RobotSDK/glviewer/CMakeLists.txt#L40](url)

Thanks.`

closed 2018-04-05 by yk-fujii

chandanm2-kpit-crpto, 2019-11-21:
hi can anybody help me regarding installation steps to be followed to install Autoware on NVIDIA Jetson TX1