autowarefoundation/autoware #289: lady bug node gets  Flip horizontal image

ohata-aki, 2016-04-27:
I found bug about "lady bug node" in "pointgrey package".

Lady bug node serve flip horizonatal images.

How about fix code as described below.

////////// original

`int main (int argc, char **argv)
{
////
cv::Mat rawImage(size, CV_8UC1, currentImage.pData + (i * size.width*size.height));
//cv::flip(mat, mat, -1);
cv::Mat image(size, CV_8UC3);
cv::cvtColor(rawImage, image, cv::COLOR_BayerBG2RGB);
cv::resize(image,image,cv::Size(size.width*image_scale/100, size.height*image_scale/100));
cv::transpose(image, image);
}`

//////// fix
`int main (int argc, char **argv)
{
/////
cv::Mat rawImage(size, CV_8UC1, currentImage.pData + (i * size.width*size.height));
cv::Mat image(size, CV_8UC3);
cv::cvtColor(rawImage, image, cv::COLOR_BayerBG2RGB);
cv::resize(image,image,cv::Size(size.width*image_scale/100, size.height*image_scale/100));
cv::transpose(image, image);
cv::flip(mat, mat, 1);
}`

Thanks.

syohex, 2016-04-27:
NOTE: re-formatting comment. Please use triple quote for code block.

syohex, 2016-04-27:
And could you show us diff file or patch file of you change or send pull request ?

ohata-aki, 2016-04-27:
Thanks for re-formatting my comment.

I will send diff file.

ohata-aki, 2016-04-27:
[diff.txt](https://github.com/CPFL/Autoware/files/238124/diff.txt)

Would that be OK with you?

syohex, 2016-04-27:
Thanks.
( However diff file generated by `git diff` is better than it. :-) )

syohex, 2016-04-27:
@amc-nu Can you know/understand this issue ?

amc-nu, 2016-04-27:
@syohex Yosh, yes I understand what he means, however I don't have a Ladybug to confirm this. @ohata-aki Ohata-san are you using a Ladybug3 or 5? Thanks for your report

ohata-aki, 2016-04-28:
@amc-nu I'm using Ladubug5. I don't confirm with Ladybug3.