autowarefoundation/autoware #291: ndt_matching step  did not work

sgzzgit, 2016-04-28:
Firstly, Thank you for answering my question!
I have download Autoware in ROS Indigo, and build it successfully. I own one velodyne HDL-32 lidar. I haved finished ndt_mapping with the lidar and Autoware already. Now i plan to run ndt_matching , but in rviz I could not see the /Points_raw frames in fixed frame /world. so I can not get the matching result. It is really annoying. I guess the problem exists in the TF. Now I have only two static publisher in TF :/world to /map , /base_link to /velodyne. Could you help me or share your Launch file. Many Thanks!

yukikitsukawa, 2016-04-29:
Did you output PCD file when finished mapping? Before runnning ndt_matching, you need to load map data(PCD file) in Map tab of Runtime Manager.
TF /map to /base_link is not static and is published in ndt_matching. You can execute ndt_matching checking the box of ndt_matching in computing tab of Runtime Manager.

sgzzgit, 2016-04-29:
Thanks for your reply very much.
My Mapping operations are list as follows:

1.Download Autoware in ROS Indigo and run RuntimeManager succeddfully.
2.Connect the hdl-32 lidar to the computer and run the RuntimeManager.
3.In 'QuickStart' --'Seneing'  ,choose 'velodyne_hdl32e,launch' and click the Sensing button.
4.In 'computing' ,click to choose 'ndt_mapping'
5.open rviz and load 'ndt_mapping.rviz'.
6.After finish mapping ,click PCDOUTPUT button to get the .PCD file.

all of above steps have been successfully operated.

To run the localization step ,My operations are as follows:
1.In 'Map' , click ref and choose the .PCD file,then click Point Cloud button.
2.After the software finished loading the .PCD file,click ref and choose tf.launch file then click on 'TF' button.
3.Because I do not have the vector map now ,so i only load these two files.
4.In 'Computing' click the 'ndt_matching'.
5.In 'Quick Start',click to start the 'sensing' to start the velodyne lidar.
5.Open the rviz ,and open the 'default.launch'. unfortunately i can see nothing in rviz.

my tf.launch file's contents are as follows:

<launch>
<node pkg="tf"  type="static_transform_publisher" name="world_to_map" args="14771 84757 -39 0 0 0 /world /map 10" />
<node pkg="tf"  type="static_transform_publisher" name="map_to_mobility" args="0 0 0 0 0 0 /map /mobility 10" />
</launch>

As it did not work correctly, so i changed the tf file as listed:

<launch>
<node pkg="tf"  type="static_transform_publisher" name="world_to_map" args="14771 84757 -39 0 0 0 /world /map 10" />
<node pkg="tf"  type="static_transform_publisher" name="base_link_to_velodyne" args="0 0 0 0 0 0 /base_link  /velodyne 10" />
</launch>

However in rviz the /Points_raw will disapper sometimes.

Thank you for solving my question! Many thanks.

yukikitsukawa, 2016-04-29:
I can't see the contents of tf.launch...

2.After the software finished loading the .PCD file,click ref and choose tf.launch file then click on 'TF' button.

I guess you selected different tf.launch. Instead of doing this, please run following command in the terminal.
rosrun tf static transform publisher 0 0 0 0 0 0 /world /map 10
Then, please confirm that pointcloud_map is shown in RViz(Open config "default.rviz").
First of all, I want to make sure that map is loaded properly.

sgzzgit, 2016-04-29:
Becaus of github's limition, I only post the major part of the tf.launch. as follows:

node pkg="tf" type="static_transform_publisher" name="world_to_map" args="0 0 0 0 0 0 /world /map 10"

node pkg="tf" type="static_transform_publisher" name="base_link_to_velodyne" args="0 0 0 0 0 0 /base_link /velodyne 10"

Now ,I can see the map which build from ndt_mapping in rviz as i launced the default.rviz. But i could not see the live velodyne data and the localization pose.

Thanks for you reply.

sgzzgit, 2016-05-01:
I have solved the problem already, Many Thanks!
I just changed the TF.launch file , and it worked.

closed 2016-05-01 by sgzzgit

elberthuang, 2017-06-02:
@sgzzgit Hello! I met the same question. How did you solve it? Thanks a lot

dlounis, 2018-01-19:
@sgzzgit  @elberthuang 
I am facing the same issue.
I do not have a lidar though. I have generated pcd submaps from moriyama sample. Then, I launched ndt_matching using the generated pcd's and live lidar scans from moriyama bag file (the same that was used to generate the pcd's).
Unfortunately I still can't display the map and the live scans on the same frame in rviz. I have modified the TF launch file in order to have the map and world frames equal (applied identiti transform). I managed to see the map on both world and map frames in rviz, however, the scans remain far from these frames (with distances over 14771 for x and 84000 for y)....

I think it has to do with the TF. Any help will be appreciated.

elberthuang, 2018-01-23:
@dlounis if you run moriyama bag file, you don't need to change anything. Just follow the steps.

dlounis, 2018-01-23:
@elberthuang Thnks for your reply. I have tried it many times with Moriyama data. If I do not use the PCD's generated by approximate_ndt_mapping (i.e. when using the provided PCD's) it works. But it has to work for PCD's the user generates.

Have you generated your own PCD's from Moriyama bag ?

shoaibazam, 2018-03-18:
@sgzzgit @kitsukawa @elberthuang  hi i have generated the map using ndt mapping on my custom data but ndt matching is not working. Both are not synchronized as in the case of demo data. Kindly help me to fix that issue.