autowarefoundation/autoware #292: How to connect Autoware to the vehicle's CAN network?

sgzzgit, 2016-05-01:
To date, I have finished ndt_mapping and ndt_matching already.
Now my next plan is to apply the localization and mission result to the vehicle. My vehicle could be driven via the can network already.

However ,I get the following questions:
1.Could Autoware output the motion planning result via can network to control the vehicle?
2.Could Autoware output the localization result via socket or serial port?
3.which type of can interface card that Autoware support? Does it support Vector VN1630A?

Thanks a lot, and if there is anything that I can do for your team ,Please contact me, I will try my best to  do it well and contribute my own strength.

sgzzgit, 2016-05-03:
Thanks for your reply.
Under your help and the operation video, now I have finished mapping, matching and wf_simulator. I am deeply grateful to you both for your help.

Now I meet one question: what should I do to connect Autoware's computing result with the real vehicle. Of course ,the vehicle could be controlled by commands via CAN network now. 

In your introduction video, Autoware could communicate with the real vehicle via CAN network, could you have an introduction of the CAN protocol and the type of the CAN interface card that you used in your vehicle? Is it "Kvaser Leaf Light V2"?

Should I write a node in ROS to subscribe the result topic in Autoware and publish the message via serial port or CAN interface card? 

In Autoware I find one folder named "vehicle", Is this folder working in the real vehicle?

Thanks in advance.

sgzzgit, 2016-05-05:
could you help me about my question? @syohex 

Thanks in advance.

shinpei0208, 2016-05-05:
Hi, Japan has long holidays now. Pls wait for a few days.

sgzzgit, 2016-05-05:
oh, I do not know that before, I am really sorry to disturb you when you are on holiday.

I will wait for your reply a few days later.

sgzzgit, 2016-05-09:
Good morning everyone, Happy Monday.

syohex, 2016-05-09:
> could you help me about my question? @syohex

No. Sorry, I don't understand this topic well.

sgzzgit, 2016-05-09:
I am sorry that I didn't describe my question well.
I want to know how to utilize the Autoware's computing result to the control the real vehicle?
Or could you tell me which one is responsible for this part in your team? @syohex 
Thanks very much.

syohex, 2016-05-09:
Can anyone answer about this issue ?

kuriking, 2016-05-09:
Autoware communicates with the real vehicle using vehicle_socket.
twist_filter publishes twist_cmd, which is subscribed by vehicle_sender.
vehicle_sender and vehicle_receiver communicate with ZMP RoboCar (Autoware/vehicle/zmp/) via socket.

sgzzgit, 2016-05-09:
Thanks for your reply!
You have solved my problem perfectly.

closed 2016-05-09 by sgzzgit

nsakar, 2016-05-09:
Can you tell me the units of measure of the twist cmd rostopic published by pure pursuit algorithm.

Is linear acceleration is meters.sev , kilonmeteres per second [ or minute, hour]?

What is the unit of measure of angular orientation velocity [ radians per sec or degree per sec]  ?

I am attempting to integrate the Autoware stack with a ros driver for another car that accepts linear x is kilometers per hor and angular velocity in radians per sec.

Regards

Nandini

h-ohta, 2016-05-09:
@nsakar 
Thank you for your question.

>Is linear acceleration is meters.sev , kilonmeteres per second [ or minute, hour]?

meter per sec .

>What is the unit of measure of angular orientation velocity [ radians per sec or degree per sec]  ?

radians per sec.

So in order to apply Autoware in your environment , please convert units of linear and angular velocity.

arashroomi, 2017-12-18:
I am looking into the vehicle/ZMP file to understand the connection of the Autoware to the vehicle and it seems that the code HevCnt.h includes a library named: "HevControl.h". But I cannot find the code for this library in any Autoware directory. Can you help me in this regard? Is it deliberately not included in the files?

dejanpan, 2017-12-25:
@arashroomi there are different ways to have Autoware to talk to different vehicles' CAN interfaces.
It depends on the drive-by-wire interface. I am aware of 2:
- pacmod from AutonmouStuff: https://github.com/astuff/ros_pacmod
- dataspeed from DataSpeed: http://wiki.ros.org/dbw_mkz_can