autowarefoundation/autoware #293: No driving_lane.csv file?

sgzzgit, 2016-05-01:
I am very sorry to bother you.
I have run the Autoware in ROS Indigo successfully, however following your instructions in the video, I could not find a lot of files that Autoware needed, like driving_lane.csv in 'waypoint_loader' and so on.
Please tell me where to get it.Thanks in advance!

sgzzgit, 2016-05-03:
Please give me a reply. Thanks in advance! It is a little urgent to me.

syohex, 2016-05-03:
@syouji @manato @h-ohta  Do you have any information ?

Can we put any sample CSV files in Autoware repository for Autoware Users ? If not, we should add documentation of creating input CSV files for them.

sgzzgit, 2016-05-03:
If you are not convenient to put any sample CSV file, Please introduce the format of the CSV files' content. I will create the CSV file myself. Is it consist of waypoints of the map?

h-ohta, 2016-05-03:
I'm sorry for late reply.

you can download example that waypoint_loader uses from "3D map of Moriyama in Nagoya", at Sample Data in Readme.
There are "moriyama_path.txt" as waypoints file in "path" directory.

This format is " x,y,z,velocity(kmph)".
x,y,z is 3-d position , velocity is target velocity.
However, you can use another format "x,y,z,yaw,velocity(kmph)".
yaw is car direction.

sgzzgit, 2016-05-03:
Thanks for your reply!

I have found the "moriyama_path.txt" file already.and  I will test it in my computer Now.

Thanks again!

sgzzgit, 2016-05-03:
I have tried following your instructions ,and now it loaded the waypoints successfully!

Many Thanks!

closed 2016-05-03 by sgzzgit