autowarefoundation/autoware #294: wf_simulator  bug  report

sgzzgit, 2016-05-03:
Firstly ,thanks for your open source code and answering my question.

I have got a bug in Autoware when I am trying to use wf_simulator in my own computer.
My operations are listed below:
1.launch Autoware and open rviz ,then open "default.rviz" in rviz.
2.select the .pcd file and click "point cloud" button.
3.select "tf.launch"  file and click the "TF" button. Now in rviz i could see the Points Map.
4.select .csv file and click "waypoint_loader" ,then I could see the waypoints in the rviz.
5.select "lane_rule" and click the "APP" of "lane_select" , and click to select "Driving".
6.click the "APP" of "wf_simulator" ,but nothing shows up.

All my operations are following the instructions in the video "Planning with wf_simulator". In the video in  step 6, there should show up a dialog to choose from "Rviz" "ndt_loclizer" "GNSS". However in my computer ,nothing shows up.

and the tips in the console is listed below:

Traceback (most recent call last):
  File "/home/linfeng/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 680, in OnHyperlinked
    self.OnHyperlinked_obj(event.GetEventObject())
  File "/home/linfeng/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 688, in OnHyperlinked_obj
    dlg = klass_dlg(self, pdic=pdic, gdic=gdic, prm=prm)
  File "/home/linfeng/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2595, in __init__
    self.panel = ParamPanel(parent, frame=frame, pdic=pdic, gdic=gdic, prm=prm)
  File "/home/linfeng/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2295, in __init__
    vp = VarPanel(self, var=var, v=v, update=self.update)
  File "/home/linfeng/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2401, in __init__
    self.choices_sel_set(v)
  File "/home/linfeng/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2572, in choices_sel_set
    self.obj.SetStringSelection(v)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_controls.py", line 2437, in SetStringSelection
    return _controls_.RadioBox_SetStringSelection(*args, **kwargs)
TypeError: String or Unicode type required

I am really sorry to disturb you, and hope my bug report could help you make Autoware much stronger.

h-ohta, 2016-05-03:
Please update latest version.
In https://github.com/CPFL/Autoware/pull/290 I fixed that bug.
In addition , you have to delete "param.yaml" in ros/src/util/package/runtime_manager/scripts/ after update.

sgzzgit, 2016-05-03:
Thank you very much! 
The Autoware works fine now.

closed 2016-05-03 by sgzzgit