autowarefoundation/autoware #295: COLLADA error when setting the vehicle model

RicWg, 2016-05-04:
Hi, thanks for such a great package.

I hit a problem when setting the vehicle model, 
I am using Ubuntu 14.04, Indigo, 
car model from :  Autoware/ros/src/.config/model/default.dae
error log is below:

thanks
Ric

['roslaunch', 'model_publisher', 'vehicle_model.launch', 'model_path:=/home/richard/Autoware/ros/src/.config/model/default.dae']
pid=8691
sched policy=OTHER prio=0
[ERROR] [1462398475.448040267]: COLLADA error: Trying to load an invalid COLLADA version for this DOM build!

[ERROR] [1462398475.468431809]: COLLADA error: Failed to load XML document from memory

[robot_state_publisher-2] process has died [pid 8714, exit code -11, cmd /opt/ros/indigo/lib/robot_state_publisher/state_publisher __name:=robot_state_publisher __log:=/home/richard/.ros/log/935206e2-1241-11e6-9ef7-782bcbabe408/robot_state_publisher-2.log].
log file: /home/richard/.ros/log/935206e2-1241-11e6-9ef7-782bcbabe408/robot_state_publisher-2*.log
Traceback (most recent call last):
File "/opt/ros/indigo/lib/joint_state_publisher/joint_state_publisher", line 343, in <module>
jsp = JointStatePublisher()
File "/opt/ros/indigo/lib/joint_state_publisher/joint_state_publisher", line 26, in __init__
robot = xml.dom.minidom.parseString(description).getElementsByTagName('robot')[0]
IndexError: list index out of range
[joint_state_publisher-1] process has died [pid 8713, exit code 1, cmd /opt/ros/indigo/lib/joint_state_publisher/joint_state_publisher __name:=joint_state_publisher __log:=/home/richard/.ros/log/935206e2-1241-11e6-9ef7-782bcbabe408/joint_state_publisher-1.log].
log file: /home/richard/.ros/log/935206e2-1241-11e6-9ef7-782bcbabe408/joint_state_publisher-1*.log
... logging to /home/richard/.ros/log/935206e2-1241-11e6-9ef7-782bcbabe408/roslaunch-deepdriving-8691.log
Checking log directory for disk usage. This may take awhile.

sgzzgit, 2016-05-05:
I met the same question before, may be you can try the ".urdf" file in the same folder with "default.dae".

RicWg, 2016-05-06:
thanks, that works,although not sure what's the difference.
are you working on reproducing the whole project?

sgzzgit, 2016-05-06:
Yes, When i watch the autoware demo video first time,I was really shocked. So i decided to devote my own strength on it.
Now my first step is to reproducing the whole project.

RicWg, 2016-05-06:
Could you page me at flyingrichard#gmail.com?
I have a bunch of guys here doing the same project. We can share some status.
thanks

closed 2016-05-06 by RicWg

sgzzgit, 2016-05-06:
Please give me a link to your page. I could not find it in yahoo.com.

RicWg, 2016-05-06:
Sorry, I mean drop an email to flyingrichard@yahoo.com

sgzzgit, 2016-05-06:
ok, we may communicate more via email.

layssi, 2016-06-15:
I don't think this works as I was able to load the model but I didn't see the model in RVIZ.