autowarefoundation/autoware #298: which type of camera does autoware use?

sgzzgit, 2016-05-12:
Thanks in advance.
In the autoware's video , It could detect the Pedestrian and traffic signal. Could you tell me how many cameras does autoware use and whta is the type of the camera?
Thanks again.

shinpei0208, 2016-05-12:
We use at least two cameras. One dedicated for traffic light recognition
and the other used for generic detection.

sgzzgit, 2016-05-12:
what is the interface of the cameras? USB?  Ethernet?

shinpei0208, 2016-05-12:
We don't select cameras. Up to your preference. You just need a driver. By
default we support drivers for PointGrey Glasshopper/LadyBug, Baumer, and
generic USB cameras such as webcams.

sgzzgit, 2016-05-12:
Thanks very much.

closed 2016-05-12 by syohex

nsakar, 2016-05-12:
How are the imaging algorithms integrated into Mission and motion planning?
Is the motion planning using the pedestrian detection or traffic light detection at all?

How does motion planning know that it needs to avoid collision with objects or stop at traffic light?

I have the following graph for enabling motion planning:

[cid:image001.jpg@01D1ABB6.832A4C80]

sgzzgit, 2016-05-12:
I could not see you graph ,could you add a link to your jpg file?

nsakar, 2016-05-12:
Please find attached graph. I would like to know how to add object recognition and traffic sign/ light detection into mission/motion planning flow.

Regards

Nandini

syohex, 2016-05-12:
> Please find attached graph.

NOTE: Email reply ignores attached file. You need to use github Web interface instead of email reply for attaching file in this issue.

manato, 2016-05-12:
@nsakar ,

>  I would like to know how to add object recognition and traffic sign/ light detection into mission/motion planning flow.

In Autoware, a node named `obj_reproj` converts detected objects position on a image into 3D position.
In other words, object's 2D coordinate on a image are transformed into 3D coordinate system by using positional relationship between camera and lidar.
Motion planning will use that 3D object coordinate so avoid collision.
(This integration is in progress and not included in master branch.)

With regard to traffic light recognition, a node in mission planning package subscribes recognition result. It switches base lane that planner tries to follow according to detected traffic light state. The difference between lane for red and green signal is only target velocity contained in waypoints around intersection.

nsakar, 2016-06-22:
Which branch has the integration of obj_reproj with motion planning is progress?