autowarefoundation/autoware #307: which camera id should I choose

sgzzgit, 2016-05-18:
I connect one usb camera to the computer.
In Aotoware "Sensing"-->Cameras , I click to choose the "USB Generic". Then in rviz I could see the raw Image of the USB camera lively.
And then In Autoware'  "Computing"-->"Detection"-->"cv_detection" , I click to choose "dpm_ocv", then one dialog appeared and asked me to choose my Camera ID. I have tried every possible option. but I could not see any detection result in rviz.

Please tell me the problem in my operations, if convenient ,Please have an introduction of "dpm_ocv" "dpm_ttic" "rcnn_node" "range_fusion" and so on.

Thanks in advance.

manato, 2016-05-18:
@sgzzgit ,

As dpm_ocv (this applies to dpm_ttic and rcnn_node too) itself calculates coordinates of detection target on a **image** (this means that result is information on 2D) , we cannot see detection result on RViz which basically shows 3D information without some information that is required by conversion of 2D to 3D.

If you simply would like to see detection result from dpm_ocv, "image_viewer" in detection tab can be helpful.
When you invoke image_viewer, two images will appear and detection result will be drawn on left side image.

I hope this will help you.

sgzzgit, 2016-05-19:
Thanks for your reply.
I invoked "image_viewer" ,However, Only one dialog appeared and there is nothing on it. I saved the screen as a picture below.

I just want to use the usb camera to capture image and see the detection result of Pedestrian and car   in Autoware.

My operations are as follows:
1.Connect one usb camera to the computer.
2. Run Autoware
3.invoke "USB generic"
4.invoke"dpm_ocv"
5.invoke "image_viewer"
Please tell me the problems in my operations. Thanks.

[image]

sgzzgit, 2016-05-19:
I guess my operations are wrong, but i am not sure.

sgzzgit, 2016-05-20:
Could you pay attention to my question? @manato .Thanks very much!

appending:
when I invoke "dpm_ocv", The dialog asked me to choose "Camera ID",I choosed "camera". Is it right?

syohex, 2016-05-20:
Did any data be published in image topic like `/image_raw` ? You can check it by `rostopic echo` etc.

sgzzgit, 2016-05-20:
I have checked the topic /image_raw , there is a lot of data in this topic.
And in rviz ,I could see the image of the camera when I invoke /image_raw topic.

sgzzgit, 2016-05-20:
if the operations are correct, Should "image_viewer" show the original image of the camera?

syohex, 2016-05-20:
No. We found issue about USB camera. Could you retry with following comment ?

```
% rosrun topic_tools relay '/image_raw' '/uvc_camera/image_raw'
```

sgzzgit, 2016-05-20:
OK, I have tried following your command , but there seems nothing changed.
Please refer to the following two images.
In rviz I could see the image lively.
[image]

sgzzgit, 2016-05-20:
In rviz I could see the image lively. "/image_raw" or '/uvc_camera/image_raw' is able to see the image.
@syohex 
[image]

syohex, 2016-05-20:
Did you get following dialog at clicking `image_viewer` checkbox ?

[image]

sgzzgit, 2016-05-20:
Thanks very much, you haved solved my problem.
I have met a lot of questions ,now I sum up in case others met the same question.

In autoware before I run the following command:
% rosrun topic_tools relay '/image_raw' '/uvc_camera/image_raw'
I can only see this if I click "image_viewer" checkbox.
[image]

So, I can not get the correct result.
After running this command, My dialog changed, and it could work well.
Thanks  again @syohex .
[image]

closed 2016-05-20 by sgzzgit

syohex, 2016-05-20:
However this is workaround. We'll fix it later.

sgzzgit, 2016-05-20:
OK ,waitting for your update.and thanks again.

manato, 2016-05-20:
@syohex , @syouji 
I'm really sorry for my late reply. 
And thank you very much for a lot of your help.

With regard to conversation of this thread, I'm sorry again to bother you, but I'm not sure why #313 was made as solution.

Camera ID that we can choose in dialog is a list of strings which is a part of particular topics.
That particular topics contains "image_raw" in its name.
When you choose a string (ex. "/camera1") as camera_ID, `image_viewer` will subscribe a topic that start with specified camera_ID (ex. `image_viewer` will subscribe "/camera1/image_raw").

On the other hand, at least in grasshopper_node, camera node will publish "/image_raw" when only one camera is detected.
And in a case that one camera is connected, `image_viewer` should display camera image lively if we specify "/" (or blank) as camera_ID (This will lead `image_viewer` to subscribe "/image_raw".).

In #313, it seems that "/image_raw" is relayed as "/camra0/image_raw".
Could you please tell my why this modification can be a solution of the problem of this thread?
(`image_viewer` or nodes using  "image_raw" as input may contain bug in some cases...)

syohex, 2016-05-23:
@manato

> Could you please tell my why this modification can be a solution of the problem of this thread?
(image_viewer or nodes using "image_raw" as input may contain bug in some cases...)

**First, #313 change is work-around. We should consider better software design.**

Runtime manger uses `/image_raw` topic by default if there is only one topic which contains `image_raw`. While runtime manager let users choose camera ID if there are more than two such topics, because runtime manager assumes that multi-camera system like ladybug is connected. A camera ID is topic name which is deleted '/image_raw' from original topic name. However multiple topics(`/image_raw`, `/image_raw/compressed` etc) which contains `/image_raw` are created when USB camera is connected even if USB camera is single camera system. Then runtime manager misrecognizes that multi-camera system is connected and let users choose camera ID. But `/image_raw` topic is removed from candidates because it becomes empty string by deleting `/image_raw` from original name. This causes this issue.

#### In japanese

Runtime managerは `/image_raw`トピックが一つしか作成されないカメラについては, それをデフォルトの入力として `image_viewer`等に表示させようとします. しかし `foobar/image_raw`のようなトピックが複数あると, それを多眼カメラと認識し `/image_raw`を取り除いたものを pull downメニューから選択させ, image_viewer等に表示させようとします. USBカメラの場合は入力が多眼カメラでない場合でも `/image_raw`の他に `/image_raw/compressed`のようなトピックも作成されてしまうため, Runtime managerが誤認識し, `/image_raw`を取り除いたものを pull downメニューに表示させます. このとき肝心の `/image_raw`は空文字列となってしまい, 候補から除外されてしまいます. これが本問題が起こった原因です.

manato, 2016-05-23:
@syohex,
Thank you very much for your reply.
Now I understand that #313 is work-around.

> However multiple topics(/image_raw, /image_raw/compressed etc) which contains /image_raw are created when USB camera is connected even if USB camera is single camera system. 

I didn't know about that!
We may have to consider not only existence of `image_raw` in topic name but also its position to show the list of camera ID correctly...

Anyway, thank you again for your answer, @syohex!