autowarefoundation/autoware #311: ring_filter build error: fatal error: velodyne_pointcloud/point_types.h: No such file or directory

RicWg, 2016-05-19:
Hi,

at the date of 05/19, the version by git cannot be compiled

[ 86%] Built target kf
Scanning dependencies of target ring_filter
[ 86%] Building CXX object sensing/data_filter/packages/points_filter/CMakeFiles/ring_filter.dir/nodes/ring_filter/ring_filter.cpp.o
~/Autoware/ros/src/sensing/data_filter/packages/points_filter/nodes/ring_filter/ring_filter.cpp:38:45: fatal error: velodyne_pointcloud/point_types.h: No such file or directory
 #include <velodyne_pointcloud/point_types.h>
                                             ^
compilation terminated.
Linking CXX executable ~/Autoware/ros/devel/lib/points_filter/distance_filter

I hacked the cmake file by adding the directory of the header file, it can work. you guys may have an more elegant way to solve it .
include_directories(include 
   ${catkin_INCLUDE_DIRS} 
   ~/Autoware/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_pointcloud/include/
)

thanks
Richard

syohex, 2016-05-20:
Did you install `ros-indigo-velodyne-pointcloud` package ?

RicWg, 2016-05-20:
I did not installed that one, as instructed
"NOTE: Please do not install ros-indigo-velodyne-pointcloud package. Please uninstall it if you already installed."

should I install  ros-indigo-velodyne-pointcloud?

syohex, 2016-05-20:
Oh, sorry for my mistake. No you should not. It is imported into Autoware repository. It looks compile flag is not collect, I'll check it.

yukikitsukawa, 2016-05-20:
Add velodyne_pointcloud to find package.
find_package(catkin REQUIRED COMPONENTS
  roscpp
  pcl_ros
  sensor_msgs
  pcl_conversions
  velodyne_pointcloud
  runtime_manager
  message_generation
)

Add "velodyne_pointcloud" to build_depend and to run_depend to package.xml as follows.
<?xml version="1.0"?>
<package>
  <name>points_filter</name>
  <version>0.0.0</version>
  <description>The points_filter package</description>
  <maintainer email="yuki@ertl.jp">Yuki KITSUKAWA</maintainer>
  <license>BSD</license>
  
  <buildtool_depend>catkin</buildtool_depend>
  
  <build_depend>sensor_msgs</build_depend>
  <build_depend>runtime_manager</build_depend>
  <build_depend>message_generation</build_depend>
  <build_depend>velodyne_pointcloud</build_depend>

  <run_depend>sensor_msgs</run_depend>
  <run_depend>message_runtime</run_depend>
  <run_depend>velodyne_pointcloud</run_depend>  
  <export>
  </export>
</package>

syohex, 2016-05-20:
@yukikitsukawa Could you send pull request ?

yukikitsukawa, 2016-05-20:
@syohex I have just sent PR.

syohex, 2016-05-20:
@RicWg We have fixed this issue at #312, please check latest version.

RicWg, 2016-05-20:
will do, light speed you guys.
thanks

closed 2017-05-29 by yk-fujii

supersampard, 2017-08-17:
I met same problem .  
[image]

I followed the kitsukawa's comment. However, I did't work
[image]
[image]

Could you explain in detail?

dejanpan, 2017-08-25:
you do not build_depend snd run_depend on `but_velodyne` pkg.