autowarefoundation/autoware #315: The protocol of vehicle_sender and vehicle_receiver

sgzzgit, 2016-05-20:
Thanks for answering my question!
Now I am trying to output the "twist_cmd" via vehicle_sender. Now I have finished connecting the computer on which Autoware is running to another computer via scoket. I received a list of data shows in the following picture.
Could you please introduce the  protocol of vehicle_sender and vehicle_receiver.
Thanks very much.

[image]

sgzzgit, 2016-05-20:
After reading the code, I understand part of the protocol. 
1.Protocol of vehicle_sender.just as shows in the following picture.
[image]

linear_x,angular_z,modeValue,gearValue,accellValue,brakeValue,steerValue.

Is it right?

2.Protocol of vehicle_receiver
Mode,Time,Veloc,Angle,Torque,Accel,Brake,Shift

Is it right?

sgzzgit, 2016-05-20:
Sorry to bother you, I met another strange problem.
When I  try to connect to "vehicle_receiver" via port 1000,The connection has been built successfully.However when I tried to connect to "vehicle_sender" via port 10001,I could get only one frame of data,then the connection  lost itself, If I reconnect it again,I could get only one frame like before and the connection lost again.
I am very confused.

syohex, 2016-05-23:
@syouji Do you have any information ?

syouji, 2016-05-23:
> linear_x,angular_z,modeValue,gearValue,accellValue,brakeValue,steerValue.
>
> Is it right?

Yes, It is.

> Mode,Time,Veloc,Angle,Torque,Accel,Brake,Shift
>
> Is it right?

No, it isn't.
key1,value1,key2,value2,...
It doesn't matter the order.
Please look at https://github.com/CPFL/Autoware/blob/master/vehicle/zmp/autoware_socket/can.cpp.

syohex, 2016-05-23:
> linear_x,angular_z,modeValue,gearValue,accellValue,brakeValue,steerValue.
> 
> Is it right?
>
> Yes, It is.

NOTE.  [Receiver code](https://github.com/CPFL/Autoware/blob/557c3aa593b72fa5e405341dec43d50b0aff0f6d/vehicle/zmp/autoware_socket/cmd.cpp#L139-L157) looks wrong. **brake** and **steer** values are reversed against sender order.

sgzzgit, 2016-05-23:
Thanks for your reply very much.

I am very confused why "vehicle_sender" will close the socket connection after sending one frame of message,I think "vehicle_sender" should send the data to the vehicle at a fixed frequency.
Thanks in advance. @syouji

syouji, 2016-05-23:
In the current implementation, a client program request "twist_cmd" from vechicle_server at intervals.                                         Sample code of client program is [this](https://github.com/CPFL/Autoware/blob/557c3aa593b72fa5e405341dec43d50b0aff0f6d/vehicle/zmp/autoware_socket/cmd.cpp#L315-L355).

sgzzgit, 2016-05-23:
OK, Thanks for answering my question.

closed 2016-05-23 by sgzzgit