autowarefoundation/autoware #319: How to split the mapping data into several small files

sgzzgit, 2016-05-26:
Thanks for your contribution first.
Now I use rosbag to record velodyne data with one real vehicle driving in the school, And run "ndt_mapping" with the simulation playing the recorded data by rosbag.

However the data is rather too big,Only part of the school occupies 40G. The memory of my workstation is 32G. So I could not get the "ndt_mapping" result.

As in your Demo data ,the mapping result had been split into many small files ,Could you tell me how to achieve that. Should I split into files when recording velodyne data with rosbag?
Thanks in advance.

syouji, 2016-05-26:
You can shrink the ndt_mapping result to inclease "Filter Resolution" at [app] dialog. Please pick somewhere around 0.2-0.5.

sgzzgit, 2016-05-26:
Thanks for your reply.
I know the method you have described above, but to my question the method is useless.
The memory of my computer is 32G,however "ndt_mapping" requires more memory.So I could not finish running "ndt_mapping". The "ndt_mapping" progress will die because there is no available memory of the computer.
In your demo data ,the map data consist of many small ".pcd" files.I want to know how to achieve that.Thanks. @syouji

sgzzgit, 2016-05-26:
Should I split into files when recording velodyne data with rosbag? @syouji

syouji, 2016-05-26:
Please wait. I'm inquiring about it.

yukikitsukawa, 2016-05-26:
The ndt_mapping can not output pcd files separately, just can output one file. So please execute ndt_mapping  several times pausing rosbag and output small pcds. Afterwards, transform each pointcloud manually and create whole map because each pcd has different origin. However, current Autoware doesn't have a function to concatenate two point clouds consistently, so please implement it by yourself. I am sorry to take you trouble.

sgzzgit, 2016-05-27:
Thanks for your reply.
I want to know how to edit the origin of one pcd file manually?

closed 2016-06-14 by sgzzgit