autowarefoundation/autoware #321: How to navigate in a larger place

sgzzgit, 2016-05-27:
Thanks for your contribution first.
Now I applied Autoware into the real vehicle. It could drive by itself using ndt_matching , waypoint_loader, pure_pursuit and so on.
However the vehicle can only drive in the area with the map built before.But to a city, it is difficult to build a map of the whole city, I know Autoware could navigate with GNSS. However part of the city is GNSS-dark areas.
So Could you tech me how to combine GNSS and 'ndt_mapping' to navigate in a larger place.
Thanks very much.

sgzzgit, 2016-05-27:
GNSS Error may be lager than 3 meters some times.

yukikitsukawa, 2016-05-27:
Currently, our system can drive within the map.
The ndt_mapping can only create local maps and can't create in global coordinate. There are no functions that combine GNSS and ndt_mapping currently.
We ask map company to provide large global maps such as whole city.

sgzzgit, 2016-05-27:
> Currently, our system can drive within the map.

Please introduce the content of the map,Is the map consist of latitude and longitude information?

shinpei0208, 2016-05-28:
Yes, and RGB information. Beyond pointcloud, it contains vector objects,
such as zebra zones, traffic lights, lanes, lines, and road networks. They
are all collections of position data (x, y, z). See:
https://youtu.be/yOEro_GChoM

Here is a link to the map company.
http://www.aisantec.co.jp/english/

sgzzgit, 2016-05-29:
Thanks very much.

closed 2016-05-29 by sgzzgit