autowarefoundation/autoware #334: How to debug latency issues between Autoware stack and car ros driver

nsakar, 2016-06-01:
Hi 

I have integrated the autoware stack with the mkz ros driver  that uses socketcan interface. 
Now the twist commands are generated and converted to CAN Commands by the ROS driver
In simulation, I can see the vehicle move perfectly in rviz.
However when actually attempting to run the physical car , there is some delay in applying teh steering commands and the car does not drive smoothly.
Please advise how to trouble shoot this issue. The ndt_pose is being computed using velodyne lidar puck that has a scan frequency of  20 z. The driver has a built in time callback where it generates the can commands from twist commands and is called 50 times per second.
I am enclosing the rosgraph
[image]

sgzzgit, 2016-06-01:
> However when actually attempting to run the physical car , there is some delay in applying teh steering commands and the car does not drive smoothly.

Does it drive smoothly in your simulation? I met the same problem as you, however in my simulation ,the car does not drive smoothly.  And in physical vehicle testing there is nearly no delay at all.

nsakar, 2016-06-01:
In simulation, it drives smoothly

Regards

Nandini

closed 2016-06-14 by nsakar

sgzzgit, 2016-06-25:
Have you solved this problem?
Did you upload the vehicle's speed to Autoware?
I am very confused by this problem.
Thank you very much.

sgzzgit, 2016-06-25:
I am sorry, I missed your description below.
> Resolved the issue by ensuring calc_twist is computed everytime even if vehicle is following trajectory

Please explain your steps detailedly. Have you changed the Autoware's source code? How to ensure calc_twist is computed?

Thanks very much. @nsakar

nsakar, 2016-06-25:
Yes. I solved this by setting vehiclFollowing flag in calcTwist to false. That in turn made the twist commands to be computed everytime. And solved my problem.

Regards

Nandini

nsakar, 2016-06-25:
Yes. I had to set vehicleFollowing to false.

Regards

Nandini

sgzzgit, 2016-06-25:
Thanks for your answer.
However I want to confine that you changed 'vehicleFollowing' or 'verifyFollowing'  in 'pure_pursuit.cpp'?

I did not found 'vehicleFollowing' in 'pure_pursuit.cpp'.
Thanks.

nsakar, 2016-06-25:
verifyFollowing sorry

sgzzgit, 2016-06-25:
[image]

Sorry to bother you again. I changed the code in line 512 of the picture above. However it did not work.
Could you tell me how to fix the code detailedly . 
Many thanks.

nsakar, 2016-06-25:
Line 512:

Bool following_flag = false;

[2]<https://cloud.githubusercontent.com/assets/4048157/16357129/a7b6f392-3b20-11e6-8bf4-3c64e622315f.JPG>

sgzzgit, 2016-06-26:
Thanks for your reply. I will fix it on my physical vehicle.

sgzzgit, 2016-06-27:
Sorry to bother you again.
I met a strange problem. After applying Autoware to the physical vehicle, the steering wheel will joggle occasionally when driving autonomously.
I check the data and find that twist data in the twist_filter not changed abruptly.However twist data in pure_pursuit changed abruptly. Then the steering wheel joggled.
Have you ever met the same problem? @nsakar

nsakar, 2016-06-27:
What velocity have you set your waypoints at in your trajectory? At higher velocities I find there is over or under steering. I am assuming  ndt matching is not fast enough to conduct localization at higher velocities. Can anyone answer the question?

sgzzgit, 2016-06-27:
Thanks for your answer.
In my trajectory, I set the velocity 3 m/s. 

> At higher velocities I find there is over or under steering. 

Could you tell me the value of the higher velocities in your description.

nsakar, 2016-06-27:
20 kph.  The  trajectory velocity is in kph.

sgzzgit, 2016-06-27:
Thank you .In my velocity set it is 10 km/h.
So I am very confused why the steering wheel joggles occasionally . And in sever tests the joggles will not repeat.

nsakar, 2016-06-27:
I am getting this issue when attempting to integrate object detection into velocity set using points_cluster.

Regards

Nandini

sgzzgit, 2016-06-27:
Have you achieved integrating object detection into velocity set using points_cluster? 
Which device you use for detecting object?
In my project,I use one camera and mm-wave radar to detect objects in front of the vehicle.

nsakar, 2016-06-27:
I use lidar and Euclidean clustering.

Have you integrated object detection with velocity_set?

sgzzgit, 2016-06-27:
I am working on it for two weeks. 
Now I havn't finished yet. After finishing ,I will share the code with you.

shinpei0208, 2016-06-28:
Euclidean clustering may not be suitable for the real time use. We often
use virtual scan.

For steering, perhaps it is a controller issue inside the physical vehicle,
not on the Autoware front. This is what often happens to us when using a
real vehicle.

sgzzgit, 2016-06-28:
Thanks for your answer. 
However I am sure the reason for the steering joggle is the pure_pursuit.cpp. I saved the data from pure_pursuit.  

> "linear : 1.587111, angular : 0.006594";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.006594";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.006594";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.007312";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.007312";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.007312";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.005560";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.005560";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.005560";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.002825";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.002825";"2";"/pure_pursuit";"
"linear : 1.587111, angular : 0.002825";"2";"/pure_pursuit";"
"linear : 1.587111, angular : -0.000251";"2";"/pure_pursuit";
"linear : 1.587111, angular : -0.000251";"2";"/pure_pursuit";
"linear : 1.587111, angular : -0.000251";"2";"/pure_pursuit";
"linear : 1.587111, angular : -0.001156";"2";"/pure_pursuit";
"linear : 1.587111, angular : -0.001156";"2";"/pure_pursuit";
"linear : 1.587111, angular : -0.001156";"2";"/pure_pursuit";
"linear : 1.587111, angular : -0.001307";"2";"/pure_pursuit";
"linear : 1.587111, angular : -0.001307";"2";"/pure_pursuit";
"linear : 1.587111, angular : -0.001307";"2";"/pure_pursuit";
"linear : 1.629667, angular : -0.000491";"2";"/pure_pursuit";
"linear : 1.629667, angular : -0.000491";"2";"/pure_pursuit";
"linear : 1.629667, angular : -0.000491";"2";"/pure_pursuit";
"linear : 1.629667, angular : -0.000025";"2";"/pure_pursuit";
"linear : 1.629667, angular : -0.000025";"2";"/pure_pursuit";
"linear : 1.629667, angular : -0.000025";"2";"/pure_pursuit";
"linear : 1.629667, angular : 0.001009";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.001009";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.001009";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.001559";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.001559";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.001901";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.001901";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.001901";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.001901";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.002591";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.002591";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.002591";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.003193";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.003193";"2";"/pure_pursuit";"
"linear : 1.629667, angular : 0.003193";"2";"/pure_pursuit";"
"linear : 1.559222, angular : 0.003772";"2";"/pure_pursuit";"
"linear : 1.559222, angular : 0.003772";"2";"/pure_pursuit";"

Please pay attention to the angular data. After two day's real vehicle tests. I am sure the problem is here. 
I am not using the latest version of autoware. my version is f58c99d. Apr 18 2016. Is my version too old? or the config parameter is wrong?
I am very confused, please help me ,Thank you.

sgzzgit, 2016-06-28:
> For steering, perhaps it is a controller issue inside the physical vehicle,
not on the Autoware front. This is what often happens to us when using a
real vehicle.

Could you tell me how you solve this problem? Adjust the config of physical vehicle?
Thank you @shinpei0208

h-ohta, 2016-06-29:
@sgzzgit 
I think we can apply lowpass filter to pure pursuit to avoid joggle steering.
I'll test this idea soon.

sgzzgit, 2016-06-30:
Great, Waitting for your test result. Please notice me if there is anything I can help you. @h-ohta

wsung1, 2018-06-20:
Hi, @nsakar!
I'd appreciate to you for sharing your valuable hands-on experience.
As you did before, now I'm trying to integrate the Autoware stack with my test vehicle. You mentioned you used the specific "ROS driver" for converting the ROS topic command (/twist_cmd) to the CAN commands. May I ask you what type of ROS driver were you depending on?
Your rqt_graph is exactly what I'd like to implement now.
[image]
Thank you very much, Nandini Sarkar :)