autowarefoundation/autoware #342: Does a stop traffic sign simply manifest itself as a reduction in twist linear velocity ?

nsakar, 2016-06-03:
Hi I am trying to implement a traffic stop sign in my motion trajectory. I expect the car to stop for some time at the defined waypoint pose and then resume.

I have studies the lane_rule and pure-pursuit algorithms. It appears the vector map signs essentially manifest themselves as a reduction in twist linear velocity at the vmap nodes. Is this a correct assumption?

Can you please explain how  a traffic sign [ not light] is integrated with pure pursuit?

sgzzgit, 2016-06-09:
Hello，Have you solved the problem yet?

nsakar, 2016-06-09:
No.. I am now able to simulate traffic light signal after finally figuring out how to build a vector map using the path’s waypoints as a baseline.

However , I do not see any nodes that would help me simulate traffic signs [ particularly stop sign] Should be similar to stop lines , however, the difference should be to stop at the stop signal for a configurable time interval and then resume.

For traffic light simulations I can do that by switching from red_waypoints_array to green_waypoints_array.

Regards

Nandini

nsakar, 2016-06-14:
yes I have

closed 2016-06-14 by nsakar