autowarefoundation/autoware #347: Install dependencies for Ubuntu 14.04 indigoの実行エラーについて

ari-trumpet, 2016-06-06:
Ubuntu 14.04LTS上で、READMEの手順に従いInstall dependencies for Ubuntu 14.04 indigoのインストールを試みました。１行目のsudo apt-get install ros-indigo-desktop-full ros-indigo-nmea-msgs ros-indigo-nmea-navsat-driver ros-indigo-sound-play ros-indigo-jsk-visualizationについてはうまく行ったのですが、２行目のsudo apt-get install libnlopt-dev freeglut3-dev qtbase5-dev libqt5opengl5-dev libssh2-1-dev libarmadillo-dev libpcap-dev gksuについて、以下のようなエラーが発生してうまく行きません。
解決法をご教示いただけないでしょうか。よろしくお願いいたします。
（エラー内容）
linux-image-3.13.0-24-generic (3.13.0-24.47) を設定しています ...
Running depmod.
update-initramfs: deferring update (hook will be called later)
Not updating initrd symbolic links since we are being updated/reinstalled 
(3.13.0-24.46 was configured last, according to dpkg)
Not updating image symbolic links since we are being updated/reinstalled 
(3.13.0-24.46 was configured last, according to dpkg)
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 3.13.0-24-generic /boot/vmlinuz-3.13.0-24-generic
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 3.13.0-24-generic /boot/vmlinuz-3.13.0-24-generic
update-initramfs: Generating /boot/initrd.img-3.13.0-24-generic
run-parts: executing /etc/kernel/postinst.d/pm-utils 3.13.0-24-generic /boot/vmlinuz-3.13.0-24-generic
run-parts: executing /etc/kernel/postinst.d/update-notifier 3.13.0-24-generic /boot/vmlinuz-3.13.0-24-generic
run-parts: executing /etc/kernel/postinst.d/zz-update-grub 3.13.0-24-generic /boot/vmlinuz-3.13.0-24-generic
/usr/sbin/grub-mkconfig: 35: /etc/default/grub: �kd�kd�kd�kd�kd�ku�ku�ku�ku�ku�ku�kb�kb�kbi�kb�kb�kb�kbiiii�kb�kb�kb�kb�kb�kb�kb�kb�kbdddddddddd�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�ku�ku�kr�kr�kr�kl�kl�kd�kset: File name too long
run-parts: /etc/kernel/postinst.d/zz-update-grub exited with return code 127
Failed to process /etc/kernel/postinst.d at /var/lib/dpkg/info/linux-image-3.13.0-24-generic.postinst line 1025.
dpkg: パッケージ linux-image-3.13.0-24-generic の処理中にエラーが発生しました (--configure):
 サブプロセス インストール済みの post-installation スクリプト はエラー終了ステータス 2 を返しました
grub-efi-amd64 (2.02~beta2-9ubuntu1.7) を設定しています ...
/var/lib/dpkg/info/grub-efi-amd64.config: 35: /etc/default/grub: �kd�kd�kd�kd�kd��kb�kb�kbi�kb�kb�kb�kbiiii�kb�kb�kb�kb�kb�kb�kb�kb�kbdddddddddd�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�ku�ku�kr�kr�kr�kl�kl�kd�kset: File name too long
dpkg: パッケージ grub-efi-amd64 の処理中にエラーが発生しました (--configure):
 サブプロセス インストール済みの post-installation スクリプト はエラー終了ステータス 127 を返しました
dpkg: 依存関係の問題により grub-efi-amd64-signed の設定ができません:
 grub-efi-amd64-signed は以下に依存 (depends) します: grub-efi-amd64 (= 2.02~beta2-9ubuntu1.7) ...しかし:
  パッケージ grub-efi-amd64 はまだ設定されていません。

dpkg: パッケージ grub-efi-amd64-signed の処理中にエラーが発生しました (--configure):
 依存関係の問題 - 設定を見送ります
dpkg: 依存関係の問題により linux-image-extra-3.13.0-24-generic の設定ができません:
 linux-image-extra-3.13.0-24-generic は以下に依存 (depends) します: linux-image-3.13.0-24-generic ...しかし:
  パッケージ linux-image-3.13.0-24-generic はまだ設定されていません。

dpkg: パッケージ linux-image-extra-3.13.0-24-generic の処理中にエラーが発生しました (--configure):
 依存関係の問題 - 設定を見送ります
dpkg: 依存関係の問題により linux-signed-image-3.13.0-24-generic の設定ができま�エラーメッセージは前の失敗から続くエラーであることを示しているので、レポートは書き込まれません。
              MaxReports にすでに達しているため、レポートは書き込まれません
                                                                           MaxReports にすでに達しているため、レポートは書き込まれません
                                                       ��ん:
 linux-signed-image-3.13.0-24-generic は以下に依存 (depends) します: linux-image-3.13.0-24-generic (= 3.13.0-24.47) ...しかし:
  パッケージ linux-image-3.13.0-24-generic はまだ設定されていません。
 linux-signed-image-3.13.0-24-generic は以下に依存 (depends) します: linux-image-extra-3.13.0-24-generic (= 3.13.0-24.47) ...しかし:
  パッケージ linux-image-extra-3.13.0-24-generic はまだ設定されていません。

dpkg: パッケージ linux-signed-image-3.13.0-24-generic の処理中にエラーが発生しました (--configure):
 依存関係の問題 - 設定を見送ります
処理中にエラーが発生しました:
 linux-image-3.13.0-24-generic
 grub-efi-amd64
 grub-efi-amd64-signed
 linux-image-extra-3.13.0-24-generic
 linux-signed-image-3.13.0-24-generic
E: Sub-process /usr/bin/dpkg returned an error code (1)

shinpei0208, 2016-06-06:
@syohex 

ari-trumpet is a student from The University of Tokyo. Can you please help him out to complete this installation. We might need to modify README? He is just following README but ROS fails to compile.

syohex, 2016-06-06:
Could you show us output of `df -h` command ? Does `/boot` partition has enough free space ?

`df -h`の結果を見せていただけないでしょうか ? `/boot`パーティションに空きはありますでしょうか ?

ari-trumpet, 2016-06-06:
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev             16G  4.0K   16G   1% /dev
tmpfs           3.2G  1.6M  3.2G   1% /run
/dev/dm-1       188G  121G   58G  68% /
none            4.0K     0  4.0K   0% /sys/fs/cgroup
none            5.0M     0  5.0M   0% /run/lock
none             16G  148K   16G   1% /run/shm
none            100M   56K  100M   1% /run/user
/dev/sda2       237M   54M  171M  24% /boot
/dev/sda1       511M  3.4M  508M   1% /boot/efi
It might be insufficient space for /boot? Could you let me know how should I do,please?

syohex, 2016-06-06:
Which kernel version do you use ? You can check it by `uname -r`.
And could you show us installed kernels in your system ? You can check it by `dpkg -l 'linux-image*'`.

ari-trumpet, 2016-06-06:
I'm using 3.13.0-24-generic.
And installed kernels are below.
un  linux-image            <なし>           <なし>           (説明 (description) がありません)
un  linux-image-3.0        <なし>           <なし>           (説明 (description) がありません)
iF  linux-image-3.13.0-24- 3.13.0-24.47     amd64            Linux kernel image for version 3.13.0 on 64 bit x
iU  linux-image-extra-3.13 3.13.0-24.47     amd64            Linux kernel extra modules for version 3.13.0 on 
ii  linux-image-generic    3.13.0.24.29     amd64            Generic Linux kernel image

syohex, 2016-06-06:
Your some kernel packages are not installed completely. (linux-image-3.13.0-24-generic and linux-image-extra-3.13.0-24-generic). Could you try following commands ?

```
% sudo apt-get install -f
```

Or

```
% sudo apt-get install --reinstall linux-image-3.13.0-24-generic and linux-image-extra-3.13.0-24-generic
```

And if states of those packages are `ii`, then colud you retry Autoware installation ?

ari-trumpet, 2016-06-06:
Neither of them seems to go well.
The result of % sudo apt-get install -f is 
linux-image-3.13.0-24-generic (3.13.0-24.47) を設定しています ...
Running depmod.
update-initramfs: deferring update (hook will be called later)
Not updating initrd symbolic links since we are being updated/reinstalled 
(3.13.0-24.46 was configured last, according to dpkg)
Not updating image symbolic links since we are being updated/reinstalled 
(3.13.0-24.46 was configured last, according to dpkg)
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 3.13.0-24-generic /boot/vmlinuz-3.13.0-24-generic
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 3.13.0-24-generic /boot/vmlinuz-3.13.0-24-generic
update-initramfs: Generating /boot/initrd.img-3.13.0-24-generic
run-parts: executing /etc/kernel/postinst.d/pm-utils 3.13.0-24-generic /boot/vmlinuz-3.13.0-24-generic
run-parts: executing /etc/kernel/postinst.d/update-notifier 3.13.0-24-generic /boot/vmlinuz-3.13.0-24-generic
run-parts: executing /etc/kernel/postinst.d/zz-update-grub 3.13.0-24-generic /boot/vmlinuz-3.13.0-24-generic
/usr/sbin/grub-mkconfig: 35: /etc/default/grub: �kd�kd�kd�kd�kd�ku�ku�ku�kb�kb�kbi�kb�kb�kb�kbiiii�kb�kb�kb�kb�kb�kb�kb�kb�kbdddddddddd�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�ku�ku�kr�kr�set: File name too long
run-parts: /etc/kernel/postinst.d/zz-update-grub exited with return code 127
Failed to process /etc/kernel/postinst.d at /var/lib/dpkg/info/linux-image-3.13.0-24-generic.postinst line 1025.
dpkg: パッケージ linux-image-3.13.0-24-generic の処理中にエラーが発生しました (--configure):
 サブプロセス インストール済みの post-installation スクリプト はエラー終了ステータス 2 を返しました
grub-efi-amd64 (2.02~beta2-9ubuntu1.7) を設定しています ...
/var/lib/dpkg/info/grub-efi-amd64.config: 35: /etc/default/grub: �kd�kd��kb�kb�kbi�kb�kb�kb�kbiiii�kb�kb�kb�kb�kb�kb�kb�kb�kbdddddddddd�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�ku�ku�kr�kr�set: File name too long
エラーメッセージは前の失敗から続くエラーであることを示しているので、レポートは書き込まれません。
                        MaxReports にすでに達しているため、レポートは書き込まれません
              MaxReports にすでに達しているため、レポートは書き込まれません
    dpkg: パッケージ grub-efi-amd64 の処理中にエラーが発生しました (--configure):
 サブプロセス インストール済みの post-installation スクリプト はエラー終了ステータス 127 を返しました
dpkg: 依存関係の問題により grub-efi-amd64-signed の設定ができません:
 grub-efi-amd64-signed は以下に依存 (depends) します: grub-efi-amd64 (= 2.02~beta2-9ubuntu1.7) ...しかし:
  パッケージ grub-efi-amd64 はまだ設定されていません。

dpkg: パッケージ grub-efi-amd64-signed の処理中にエラーが発生しました (--configure):
 依存関係の問題 - 設定を見送ります
dpkg: 依存関係の問題により linux-image-extra-3.13.0-24-generic の設定ができません:
 linux-image-extra-3.13.0-24-generic は以下に依存 (depends) します: linux-image-3.13.0-24-generic ...しかし:
  パッケージ linux-image-3.13.0-24-generic はまだ設定されていません。

dpkg: パッケージ linux-image-extra-3.13.0-24-generic の処理中にエラーが発生しました (--configure):
 依存関係の問題 - 設定を見送ります
dpkg: 依存関係の問題により linux-signed-image-3.13.0-24-generic の設定ができません:
 linux-signed-image-3.13.0-24-generic は以下に依存 (depends) します: linux-image-3.13.0-24-generic (= 3.13.0-24.47) ...しかし:
  パッケージ linux-image-3.13.0-24-generic はまだ設定されていません。
 linux-signed-image-3.13.0-24-generic は以下に依存 (depends) します: linux-image-extra-3.13.0-24-generic (= 3.13.0-24.47) ...しかし:
  パッケージ linux-image-extra-3.13.0-24-generic はまだ設定されていません。

dpkg: パッケージ linux-signed-image-3.13.0-24-generic の処理中にエラーが発生しました (--configure):
 依存関係の問題 - 設定を見送ります
処理中にエラーが発生しました:
 linux-image-3.13.0-24-generic
 grub-efi-amd64
 grub-efi-amd64-signed
 linux-image-extra-3.13.0-24-generic
 linux-signed-image-3.13.0-24-generic
E: Sub-process /usr/bin/dpkg returned an error code (1)

And the error below appeared for 
% sudo apt-get install --reinstall linux-image-3.13.0-24-generic and linux-image-extra-3.13.0-24-generic

E: Internal Error, No file name for linux-image-3.13.0-24-generic:amd64

syohex, 2016-06-06:
Could you try this one ?

```
% sudo dpkg --configure -a
```

And could you check package states again(`dpkg -l 'linux-image*'`) ?

ari-trumpet, 2016-06-06:
It seems the same error.

Failed to process /etc/kernel/postinst.d at /var/lib/dpkg/info/linux-image-3.13.0-24-generic.postinst line 1025.
dpkg: パッケージ linux-image-3.13.0-24-generic の処理中にエラーが発生しました (--configure):
 サブプロセス インストール済みの post-installation スクリプト はエラー終了ステータス 2 を返しました
grub-efi-amd64 (2.02~beta2-9ubuntu1.7) を設定しています ...
/var/lib/dpkg/info/grub-efi-amd64.config: 35: /etc/default/grub: �kd�kd�kd�kd�kd�ku�ku�ku�ku�ku�ku�kd�kd�kd�ku�ku��kb�kb�kbi�kb�kb�kb�kbiiii�kb�kb�kb�kb�kb�kb�kb�kb�kbdddddddddd�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kb�kbset: File name too longr�kr�kl�kl�kd�kd
dpkg: パッケージ grub-efi-amd64 の処理中にエラーが発生しました (--configure):
 サブプロセス インストール済みの post-installation スクリプト はエラー終了ステータス 127 を返しました
dpkg: 依存関係の問題により grub-efi-amd64-signed の設定ができません:
 grub-efi-amd64-signed は以下に依存 (depends) します: grub-efi-amd64 (= 2.02~beta2-9ubuntu1.7) ...しかし:
  パッケージ grub-efi-amd64 はまだ設定されていません。

dpkg: パッケージ grub-efi-amd64-signed の処理中にエラーが発生しました (--configure):
 依存関係の問題 - 設定を見送ります
dpkg: 依存関係の問題により linux-image-extra-3.13.0-24-generic の設定ができません:
 linux-image-extra-3.13.0-24-generic は以下に依存 (depends) します: linux-image-3.13.0-24-generic ...しかし:
  パッケージ linux-image-3.13.0-24-generic はまだ設定されていません。

dpkg: パッケージ linux-image-extra-3.13.0-24-generic の処理中にエラーが発生しました (--configure):
 依存関係の問題 - 設定を見送ります
dpkg: 依存関係の問題により linux-signed-image-3.13.0-24-generic の設定ができません:
 linux-signed-image-3.13.0-24-generic は以下に依存 (depends) します: linux-image-3.13.0-24-generic (= 3.13.0-24.47) ...しかし:
  パッケージ linux-image-3.13.0-24-generic はまだ設定されていません。
 linux-signed-image-3.13.0-24-generic は以下に依存 (depends) します: linux-image-extra-3.13.0-24-generic (= 3.13.0-24.47) ...しかし:
  パッケージ linux-image-extra-3.13.0-24-generic はまだ設定されていません。

dpkg: パッケージ linux-signed-image-3.13.0-24-generic の処理中にエラーが発生しました (--configure):
 依存関係の問題 - 設定を見送ります
処理中にエラーが発生しました:
 linux-image-3.13.0-24-generic
 grub-efi-amd64
 grub-efi-amd64-signed
 linux-image-extra-3.13.0-24-generic
 linux-signed-image-3.13.0-24-generic

% sudo dpkg --configure -a

un  linux-image                      <なし>                <なし>                (説明 (description) がありません)
un  linux-image-3.0                  <なし>                <なし>                (説明 (description) がありません)
iF  linux-image-3.13.0-24-generic    3.13.0-24.47          amd64                 Linux kernel image for version 3.13.0 on 64 bit x86 SMP
iU  linux-image-extra-3.13.0-24-gene 3.13.0-24.47          amd64                 Linux kernel extra modules for version 3.13.0 on 64 bit x86 SMP
ii  linux-image-generic              3.13.0.24.29          amd64                 Generic Linux kernel image

syohex, 2016-06-06:
**It looks your package system is broken. I recommend you to reinstall OS if possible**

Could you try to remove those packages and try again ?
```
% sudo dpkg --remove linux-image-3.13.0-24-generic  linux-signed-image-3.13.0-24-generic
```

ari-trumpet, 2016-06-07:
Not being able to go well, I reinstalled OS and succeeded to install ROS. Thank you for helping out.

syohex, 2016-06-07:
You are welcome.

closed 2016-06-07 by syohex