autowarefoundation/autoware #357: How to implement vehicle following and vehicle passing using Autoware stack

nsakar, 2016-06-13:
I would like to implement vehicle following where the car as a robot reduces speed depending on moving car in front. What ros nodes will help implement this?

I would like to also implement vehicle passing wherein if there is a stationary object on trajectory, then car will pass it. How do I implement this?

sgzzgit, 2016-06-14:
Is your testbed a real vehicle?

nsakar, 2016-06-14:
yes

sgzzgit, 2016-06-16:
I am trying to achieve that too.However there is still no progress.
Anyone could answer this question ? @shinpei0208 
Thanks in advance.

sgzzgit, 2016-06-19:
Anyone concerns this question?

nsakar, 2016-06-19:
I would like an answer to the question.

Regards

Nandini

sgzzgit, 2016-07-08:
It seems that you have solved this problem,is it? @nsakar

nsakar, 2016-07-08:
Vehicle following and collision avoidance using vscan points : Yes

Vehicle passing : No. Need more information.

sgzzgit, 2016-07-09:
You detected obstacle using velodyne vlp-16?
I am trying to finish collision avoidance using ifv250. Because in my real test, Velodyne lidar could not detect pedestrian and bicycle well. what is the performance in your test?

nsakar, 2016-07-09:
Yes. I  am able to detect all obstacles that are within detection range of waypoints and within height interval [ hc – hf]
I need more info on the ceiling and floor height settings. Is this from the ground or from the localizer pose?
By default it set to +2 , -2. Is this in meters and from ground plane as 0? Or localizer pose plane as 0? I am presuming the heights are on the z axis.

thomas3016, 2016-07-10:
Hi @nsakar ,I read the source code of vscan, I don't understand the meaning of rotlength,rotheight,and maxBackDistance and svsback,can you help me? Thank you for advance!
Regards

shinpei0208, 2016-07-10:
@alexanderhmw, are you online?

alexanderhmw, 2016-07-10:
Hi Kato-sensei , i am traveling now. I will add comment to vscan soon after getting back

alexanderhmw, 2016-07-12:
> I need more info on the ceiling and floor height settings. Is this from the ground or from the localizer pose?
By default it set to +2 , -2. Is this in meters and from ground plane as 0? Or localizer pose plane as 0? I am presuming the heights are on the z axis.

Please check https://github.com/CPFL/Autoware/issues/394. Yes the heights are on the z axis.

> I don't understand the meaning of rotlength,rotheight,and maxBackDistance and svsback,can you help me? 

rot* is the corresponding length and height after rotating the coordinate (Fig. 3 right) for downslope obstacle detection.

maxBackDistance and svsback are kind of patch to solve the overhung obstacle bug. They will keep the extension of road surface from the interruption of near overhung obstacles.

closed 2016-07-13 by nsakar