autowarefoundation/autoware #363: Using euclidean clustering to detect obstacle and impact motion

nsakar, 2016-06-22:
Is there a way to only use Euclidean clustering to detect object and impact motion planning in velocity_set? That is decelerate when object in front and within detection range and stop if necessary.?

TomohitoAndo, 2016-06-24:
> That is decelerate when object in front and within detection range and stop if necessary.?

Yes, `velocity_set` node detects objects by counting vscan points in detection range, and if the number of points is more than `Points Threshold`, the vehicle decelerates or stops.

>  If I remap cluster_points to vscan_points can I reuse the object detection impact on motion planning?

Yes, I suppose you can detect objects by modifying subscribing topic or relaying topic.
But since points published by `euclidean_cluster` node contains the vehicle roof, and the number of points is larger than vscan points, maybe velocity_set does not work well without modifying the code.

nsakar, 2016-06-25:
So remapping points_cluster to vscan_points does decelerate the vehicle near the obstacle. However the deceleration is from a certain value down to the decelerated value without a graceful gradient decsent. This manifests itself in erroneous steering computation by car can driver. How can we ensure graceful deceleration like what is done in lane_rule.cpp... The velocities on waypoints computed is gradual for a detected stop line.

nsakar, 2016-07-07:
I have been successful in collision avoidance and vehicle following using VSCAN points

closed 2016-07-07 by nsakar

sgzzgit, 2016-07-08:
collision avoidance? 
You mean when obstacle in front of the vehicle, the vehicle speed down  or pass the obstacle by planning and following a new route?
I am very interested in it. Thank you very much.

nsakar, 2016-07-08:
Yes. Using vscan points I was able to detect obstacles and velocity_set node appropriately factors in impact on motion planning.

Regards

Nandini

urbansound8K, 2022-05-10:
do you have an idea how to send the information of detected objects via udp? what I am trying to do is that I am running my velodyne 32 and applying euclidean clustering. Here, I want to send the data I can see it in RVIZ via UDP! like the distance and width