autowarefoundation/autoware #375: DPM TTIC Object Detection and Classification issues

nsakar, 2016-07-02:
I have created a rosbag using persons and car images from KITTI dataset and am using the DPM TTIC algorithm to detect and recognize the objects using the pre-trained models.

I find the results are not  correct particularly for pedestrian.. What configuration parameters can be tweaked to make the detection and recognition more reasonable?

manato, 2016-07-02:
@nsakar ,
As DPM evaluates target existence by threshold judgement, the simplest way to tweak detection result is changing threshold parameter.
`Compuging` tab -> `[app]` of dpm_ttic -> `parameter tuning for pedestrian` -> and tweak `Score Threshold`

If Score Threshold gets bigger, DPM shows much rectangles (this means false positive increase).
And, vice versa, if it gets smaller, DPM shows less rectangles (this means false negative increase).

manato, 2016-07-02:
Sorry, I made mistake about relationship between threshold and result...

If Score Threshold gets bigger DPM shows **less** rectangles (this means false **negative** increase).
And if it gets smaller, DPM shows **much** rectangles (this means false **positive** increase).

closed 2016-07-02 by nsakar