autowarefoundation/autoware #377: Points2Vscan

nsakar, 2016-07-02:
How do I invoke just the points2vscan node to generate vscan tostopic from lidar points_raw? I need to use vscan_points for obstacle detection in velocity_set, but do not have a camera.

manato, 2016-07-02:
You can invoke points2vscan node by:
`Sensing` tab -> `Virtual Scan Image`

nsakar, 2016-07-02:
Is it possible to simply view the vscan points without a camera?  I can see that /vscan_points are being generated when I invoke the following:

rosrun points2image points2vscan

manato, 2016-07-02:
As `/vscan_points` are the same data type as `/points_raw`, 
you can see them on RViz by just changing displayed data.

nsakar, 2016-07-02:
Yes. I just realized that.. ☺ Thank you…

One more question. I will be using the vscan_points to detect obstacles for my velocity_set node…

I see a bunch of parameters hard coded into the points2vscan node. Will it be sufficient to configure the velocity_set node using /config/velocity_set ros topic for my roof top lidar to detect obstacles?

Regards

Nandini

manato, 2016-07-02:
Parameters coded in points2vscan and listed in `/config/velocity_set` are completely different things.
Currently, parameters in points2vscan can be tweaked only from launch file (`Autoware/ros/src/util/packages/runtime_manager/scripts/vscan.launch`) and there is no GUI for tweaking them.
They configure the behavior of vscan_points themselves (Points on obstacle will be published as vscan_points or not).
On the other hand, parameters in `/config/velocity_set` are used in velocity_set node and they configure how vscan_points are treated (velocity_set node will recognize each vscan_points as obstacle or not).

As we usually configure only `/config/velocity_set` to detect obstacle by vscan, I think it will be sufficient to configure velocity_set to detect obstacles (at least current Autoware).

nsakar, 2016-07-02:
Thank you. Will test on vehicle

closed 2016-07-02 by nsakar

nsakar, 2016-07-03:
Can you please explain the algorithm used to compute vscan points from points_raw? How is the down sampling done and is it lossless from an obstacle detection perspective?

Regards

Nandini

alexanderhmw, 2016-07-05:
Hi Nandini,

Our algorithm refers to the paper "robust virtual scan for obstacle
detection in Urban environments" IEEE Intelligent Vehicles Symposium 2016.
Please check it for more details.

Best,
HMW-Alexander