autowarefoundation/autoware #386: my_map.launch file created by my_launch script does not complete properly

distractal, 2016-07-05:
Hi,

I'm excited to test this software.

I'm following the Quick Start video on the Autoware GitHub landing page:
https://www.youtube.com/watch?v=Ursbp2qs1u0

I got the sample data OK, and got the my_launch script and ran it in ~/Autoware.

Following along with the video, everything is OK until the Map button is clicked.  When this happens, the "load files" portion finishes but it seems to stop on "start publish vector map".

Additionally, I see only an empty grid in RViz and the error message:
"No tf data.  Actual error: Fixed Frame [map] does not exist."

I had to create a symlink from ~/.autoware to ~/Autoware to allow me to get this far, otherwise I get a "No such file or directory"
I also loaded the .urdf file instead of the .dae file for the vehicle model as I received a COLLADA error.

Ubuntu 14.04 LTS
ASUS ROG Laptop w/ NVIDIA GTX970M, 32 GB

thomas3016, 2016-07-09:
Hi friend,Have you soved this problem?,I met the same problem.

distractal, 2016-07-11:
Hi,

Here is what I have found to work.  Note that there is still some instability with the simulation playback... for around the first 60 seconds, the LIDAR location does not appear to be synchronized with the GPS location.  Additionally, intermittently it seems as though the ROSBAG feed will 'desync' with RViz, and the playback within RViz will stop and then start up again.  I say intermittent because sometimes I can play the entire sensor recording without any issues, other times it desyncs.

A Quick Start PDF was released last week that really helped.

This procedure assumes you have followed the Quick Start PDF instructions (https://github.com/CPFL/Autoware/blob/master/docs/jp/autoware-quickstart.pdf) for installation and where to place files, and appropriate files are selected for Map, Localization and Simulation portions as per the same document.

1.  Start Runtime manager (~/Autoware/ros/run)
2.  Ensure appropriate files are selected as per the Quick Start PDF.
2.  Simulation:
	Start ROSBAG simulation playback, wait 3 seconds and then pause
3.  Quick Start:
	Start map, wait for "OK" to appear next to Ref button
	Start localization, wait 10 seconds
4.  Start RViz
	Open ~/Autoware/ros/src/.config/rviz/default.rviz
5.  Return to Runtime Manager
	Simulation:
	Unpause playback for 5 seconds, then pause again
6.  Return to RViz, uncheck Vector Map and recheck Vector Map
	In RViz, uncheck Points Raw and recheck Points Raw
	On the right hand side, change Target Frame field to base_link and click the Zero button above
7.  Return to Runtime Manager
	Unpause the simulation
8.  Return to RViz and observe playback

closed 2016-07-11 by distractal