autowarefoundation/autoware #392: Why we need mapping in the slam?

thomas3016, 2016-07-09:
I know that through matching  between frames we can get the relative displacement,so we can locate our position,but I confuse why we need mapping,what are the function of the mapping?,Any help will be appreciated!
Regards

YukiKitsukawa, 2016-07-09:
In Autoware, localization(ndt_matching) and mapping(ndt_mapping) are completely separate functions. ndt_mapping is a function that create a map in a unknown environment from scratch so you don't need a map. On the other hand, ndt_matching is a localization function within a map so map is needed as a input of the function.

thomas3016, 2016-07-10:
Thank you for your reply, please forgive me my stupid.I confused that we can locate our position just through relative displacement previous between frame and current frame if we know the start position, why we need match the current frame with the map,and how to match the current frame with the established map?,what is the principle of location in ndt_mapping and ndt_matching? Thank you in advance!
Regards

YukiKitsukawa, 2016-07-11:
The reason why we need to match the scan and the existing map is it is faster and more accurate than matching the current scan and the previous scans. Simultaneously localize and mapping using NDT can not be done in real-time so far. If you want to know about NDT algorithm, please refer to the following paper.
http://ieeexplore.ieee.org/xpl/freeabs_all.jsp?arnumber=4058864&abstractAccess=no&userType=inst

thomas3016, 2016-07-11:
Thank you for you help!

closed 2016-07-11 by thomas3016