autowarefoundation/autoware #394: VSCAN min and max heights and velocity_set min and max detection heights

nsakar, 2016-07-09:
The VSCAN ceiling and floor values : Are they wrt to ground or wrt localizer pose?
Also the config values for min(detection_height) ad max (detection_height) in velocity set config are wrt localizer pose or wrt ground values?
I did study the vscan publication at  https://www.researchgate.net/publication/299616531_Robust_Virtual_Scan_for_Obstacle_Detection_in_Urban_Environments
but am confused on how the measurements are taken. Also what does beamnum signify? I have a 16 ring lidar. Does beamnum coincide with ring number or does it have some other significance?

TomohitoAndo, 2016-07-12:
> The VSCAN ceiling and floor values : Are they wrt to ground or wrt localizer pose?

They are wrt localizar pose.

> Also the config values for min(detection_height) ad max (detection_height) in velocity set config are wrt localizer pose or wrt ground values?

They are also wrt localizer pose.

> I did study the vscan publication at https://www.researchgate.net/publication/299616531_Robust_Virtual_Scan_for_Obstacle_Detection_in_Urban_Environments
but am confused on how the measurements are taken. Also what does beamnum signify? I have a 16 ring lidar. Does beamnum coincide with ring number or does it have some other significance?

@alexanderhmw Can you answer this question?

alexanderhmw, 2016-07-12:
The VScan's ceiling and floor values are wrt Velodyne pose (aka localizer).
The measurements are based on the flood-filled method filtering road surface and getting obstacles.
The beamnum defined how many beams you need in the 2D VScan.
16-ring is a kind of vertical resolution, and the beamnum is related to a kind of horizontal resolution.

closed 2016-07-13 by nsakar