autowarefoundation/autoware #397: installing problem

yunxiaoshan, 2016-07-15:
apt-xapian-index : Depends: python:any (>= 2.7.1-0ubuntu2)
 bzr : Depends: python:any
 compizconfig-settings-manager : Depends: python:any (>= 2.7.1-0ubuntu2)
 duplicity : Depends: python:any (>= 2.7.1-0ubuntu2)
 gconf2 : Depends: python:any
 hplip-data : Depends: python:any (>= 2.7.1-0ubuntu2)
 ibus : Depends: python:any (>= 2.7.1-0ubuntu2)
 ibus-pinyin : Depends: python:any (>= 2.7.1-0ubuntu2)
 ibus-table : Depends: python:any (>= 2.7.1-0ubuntu2)
 landscape-client-ui-install : Depends: python:any
 libboost-mpi-python1.54.0 : Depends: python:any (>= 2.7.5-5~)
 libglib2.0-dev : Depends: python:any (>= 2.7.1-0ubuntu2)
 mercurial-common : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-apt : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-aptdaemon : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-bson : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-bzrlib : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-cairo : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-catkin-pkg : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-chardet : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-commandnotfound : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-configobj : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-crypto : Depends: python:any (>= 2.7.5-5~)
 python-cupshelpers : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-dateutil : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-dbus : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-debian : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-debtagshw : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-defer : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-dev : Depends: python (= 2.7.5-5ubuntu3) but 2.7.12-1 is installed
 python-dirspec : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-docutils : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-empy : Depends: python:any (>= 2.7.5-5~)
               Depends: python:any (< 2.8)
 python-gi : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-gobject-2 : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-gpgme : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-gridfs : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-gtk2 : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-httplib2 : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-ibus : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-imaging : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-keyring : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-launchpadlib : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-lazr.restfulclient : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-lazr.uri : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-ldb : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-libxml2 : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-lockfile : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-lxml : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-nose : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-notify : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-numpy : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-oauth : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-oauthlib : Depends: python:any (>= 2.7.5-5~)
                   Depends: python:any (< 2.8)
 python-oneconf : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-opencv : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-opengl : Depends: python:any (< 2.8)
                 Depends: python:any (>= 2.7.5-5~)
                 Depends: python-ctypes
 python-openssl : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-paramiko : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-pexpect : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-pil : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-piston-mini-client : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-pkg-resources : Depends: python:any (>= 2.7)
                        Depends: python:any (< 2.8)
 python-psutil : Depends: python:any (>= 2.7.5-5~)
 python-pydot : Depends: python:any (>= 2.7.5-5~)
                Depends: python:any (< 2.8)
 python-pygments : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-pymongo : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-pyparsing : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-pyside.qtcore : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-qt4 : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-qwt5-qt4 : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-reportlab : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-requests : Depends: python:any (< 2.8)
                   Depends: python:any (>= 2.7.5-5~)
 python-roman : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-rosdep : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-rosdistro : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-rosinstall : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-rospkg : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-samba : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-secretstorage : Depends: python:any (>= 2.7.5-5~)
                        Depends: python:any (< 2.8)
 python-serial : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-setuptools : Depends: python:any (>= 2.7)
                     Depends: python:any (< 2.8)
 python-simplejson : Depends: python:any (>= 2.7.5-5~)
 python-sip : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-sip-dev : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-six : Depends: python:any (>= 2.7.5-5~)
              Depends: python:any (< 2.8)
 python-smbc : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-tdb : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-twisted-core : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-twisted-web : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-tz : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-ubuntu-sso-client : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-urllib3 : Depends: python:any (>= 2.7.5-5~)
                  Depends: python:any (< 2.8)
 python-vcstools : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-wadllib : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-wstool : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-wxgtk2.8 : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-wxversion : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-xapian : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-xdg : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-yaml : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-zeitgeist : Depends: python:any (>= 2.7.1-0ubuntu2)
 python-zope.interface : Depends: python:any (>= 2.7.1-0ubuntu2)
 sessioninstaller : Depends: python:any (>= 2.7.1-0ubuntu2)
 software-center : Depends: python:any (>= 2.7.1-0ubuntu2)
 software-center-aptdaemon-plugins : Depends: python:any (>= 2.7.1-0ubuntu2)
 totem : Depends: python:any (>= 2.7.1-0ubuntu2)
 totem-plugins : Depends: python:any (>= 2.7.1-0ubuntu2)
 ubuntu-sso-client-qt : Depends: python:any (>= 2.7.1-0ubuntu2)
 ubuntu-system-service : Depends: python:any (>= 2.7.1-0ubuntu2)
 unity : Depends: python:any
 what does that mean? I have installed python 2.7.12.

syohex, 2016-07-16:
I think your package management system is broken. This is not Autoware issue. I suppose you can find any hint by googling your error message.

Which Ubuntu version did you use ? Could you show us output of `uname -a` ?

>  I have installed python 2.7.12.

This looks strange. Is it Ubuntu package ? Even latest Ubuntu 16.04 provides Python 2.7.11. Did you install Python from source code ?

Could you try again following commands and retry ? If you still fail to install packages, I recommend you to reinstall OS.

```
sudo apt-get install -f
```

If you still fail, try following commands

```
% sudo dpkg --purge --pending 
% apt-get install -f 
```

yunxiaoshan, 2016-07-16:
I am using ubuntu14.04, I install python from source, I just contact
python. I try to do as you said, but it does not work. and give the error:
"E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused
by held packages.
E: Unable to correct dependencies "

uname -a
Linux bran 3.19.0-58-generic #64~14.04.1-Ubuntu SMP Fri Mar 18 19:05:43 UTC
2016 x86_64 x86_64 x86_64 GNU/Linux

yunxiaoshan, 2016-07-16:
I think you are right, it is not the problem of autoware.

2016-07-15 21:43 GMT-04:00 yunxiao shan <yunxiao.shan@gmail.com>:

closed 2017-05-29 by yk-fujii