autowarefoundation/autoware #398: can't load vehicle model

dingyanchao, 2016-07-17:
I have followed the Quick Start in Instruction Videos. When the Autoware starts to run, there is something wrong with vehicle model.The vehicle model path is "Autoware/ros/src/.config/model/default.dae". The error is "[ERROR] [1468713326.197371983]: COLLADA error: Trying to load an invalid COLLADA version for this DOM build!
[ERROR] [1468713326.254143386]: COLLADA error: Failed to load XML document from memory".
Please tell me how to run it correctly. Thank you.

YukiKitsukawa, 2016-07-18:
Can you replace "Autoware/ros/src/.config/model/default.dae" with "Autoware/ros/src/.config/model/default.dae"?

dingyanchao, 2016-07-18:
Thannks for your comment!
I'm afraid I don't quite understand what you mean.The vehicle model path  used absolute path as "/home/work-2016/Autoware/ros/src/.config/model/default.dae".

YukiKitsukawa, 2016-07-19:
I am sorry, it's a typo. Please select "Autoware/ros/src/.config/model/default.urdf".

dingyanchao, 2016-07-19:
I select "/home/work-2016Autoware/ros/src/.config/model/default.urdf",then there is no error message.
I followed the Quick Start in Instruction Videos.After select Map->select Localization->select Misssion Planning->select MotionPlanning ,then click Rviz button.The Display in Rviz GUI only have “Global Options”/"Global Status"/"Grid".There no other items in "Display" like the Quick Start Video's show.It seems that select  "vehicle model"/"Map"/"Localization"/"Misssion Planning"/"MotionPlanning" can't work correctly.  I have no idea what the problem is.Do you have any idea about the problem? thank you

dingyanchao, 2016-07-19:
I select "/home/work-2016/Autoware/ros/src/.config/model/default.urdf",then there is no error message.
I followed the Quick Start in Instruction Videos.After select Map->select Localization->select Misssion Planning->select MotionPlanning ,then click Rviz button.The Display in Rviz GUI only have “Global Options”/"Global Status"/"Grid".There no other items in "Display" like the Quick Start Video's show.It seems that select "vehicle model"/"Map"/"Localization"/"Misssion Planning"/"MotionPlanning" can't work correctly. I have no idea what the problem is.Do you have any idea about the problem? thank you

YukiKitsukawa, 2016-07-19:
Open the RViz config "Autoware/ros/src/.config/default.rviz" from "File" menu of RViz.

closed 2017-05-29 by yk-fujii