autowarefoundation/autoware #401: Re producing Quick Start video

vlnguyen92, 2016-07-27:
I got Autoware to start up, downloaded and extracted the `my_launch.sh` file to get all the `*.launch` files. 

According to the video, here are the steps:

1. Load ROSBAG under the Simulation Tab
2. Hit Play and then immediately Pause
3. Go to Setup tab and load Vehicle Model (the `default.dae` file)
3. Load my_map.launch using the `Ref` button 
4. Hit the `Map` button

The video shows that it should start loading the `*.pcd` files. But I got the following error in the `runtime_manager` terminal:

```
[ERROR] [1469589150.392194843]: Usage:
[ERROR] [1469589150.392265905]: rosrun map_file points_map_loader noupdate [PCD]...
[ERROR] [1469589150.392289088]: rosrun map_file points_map_loader {1x1|3x3|5x5|7x7|9x9} AREALIST [PCD]...
[ERROR] [1469589150.392309561]: rosrun map_file points_map_loader {1x1|3x3|5x5|7x7|9x9} download
```
How do I fix this?

kuriking, 2016-07-27:
Please try the following steps:

1. Load ROSBAG under the Simulation Tab
2. Hit Play and then immediately Pause
3. Go to Quick Start tab and hit the `Map` button
4. If you want to do ndt_matching, hit the `Localization` button
5. Hit the `RViz` button and load "default.rviz"
6. Go back to Simulation tab and hit `Pause` button

vlnguyen92, 2016-07-28:
At step 3 on your suggestions, I loaded the `my_map.launch` and then hit the `Map` button. The runtime manager gives me the same errors as in my original question.

Is there anything else I need to do before that?

closed 2016-07-29 by vlnguyen92

nmerrill67, 2017-02-15:
vlnguyen92 have you fixed this problem? If so how? I am running into the same issue, and it is very frustrating.

pxlong, 2017-08-08:
@nmerrill67 have you tried to source the Autoware/ros/devel/setup.bash (or setup.zsh if you use zsh) in your .bashrc ?