autowarefoundation/autoware #404: Hi , How can i run the traffic light detection  as stand alone application , My intention is only to test  traffic light APP

anooppottammal, 2016-07-29:
I have the software downloaded and installed  in Ubuntu machine e.. I was trying to run just the region_tlr.cpp file using GCC compiler , its shown many issue in library headers and pointers used in code.
Please help me if any one had already made the traffic light as a stand alone application , My experience in Linux and C++ is very less.

manato, 2016-08-09:
@anooppottammal 
I'm terribly sorry for late reply.

As region_tlr highly depends on other function in Autoware, I think it is hard to use it as stand alone application.

Moreover,  Autoware is composed on ROS, which has own compile system called "catkin_make".
catkin_make uses Cmake in the background in order to create makefile suited for each environment.
From this point of view, it is hard to compile region_tlr by gcc as stand alone app.

Best Regards

anooppottammal, 2016-08-09:
Ok , I got you !!
Thanks you very much for your support and I appreciate the work 👍🏻

--Anoop Pottammal

closed 2017-05-29 by yk-fujii