autowarefoundation/autoware #409: Lane change , Lane Merage and left , right turn signals

nsakar, 2016-08-03:
Can you please provide guidance on how to invoke lane change , lane merge and turn signals using the stack?

sgzzgit, 2016-08-03:
you sure autoware support Lane change now?
Have you achieved real-time Lane detection using camera on your vehicle?
Thanks very much.

h-ohta, 2016-08-04:
@nsakar Sorry for late reply.
Now in Autoware lane change is manual.
If you load two lanes at `waypoint_loader`, `lane_select` can switch lane from loaded lanes by using button `Driving` and `Passing` in app tab.

When lane is switched, path following node can follow the lane, like lane merge.

And, lane has left and right turn information.
We can turn right or left so that path follow node also follows the information.

closed 2016-08-20 by nsakar