autowarefoundation/autoware #415: Traffic Light  Feature Enablement Help

nsakar, 2016-08-05:
I have a dilemma in understanding the integration of motion panning with traffic light detection.
1. Vector map elements of relevance include signaldata.csv, vector.csv, point.csv. Point in world coordinates denoting signal center is transformed from world coordinates to image plane. Radius of signal is computed as distance  between signal center and a point = z+SignaLLampRadius. The orientation of signal is computed as signal.hang + 180.  How is  hang and vang computed? from which axis. I am presuming we are only interested in signals that are +-70 degree from camera orientation. These signals of relevance are then published as a roi signal array [ signalId,u,v,radius,x,y,z,hang,type,linkId,plid]

2.The region_tlr node detects signal state at the roi signal locations. 

Please provide some guidance on how to set this all up correctly.

Regards

2. Only signals that meet the constraint are candidates for traffic light detection

manato, 2016-08-06:
@nsakar ,
> How is hang and vang computed?

In Autoware, we get hang  by following:
 - Get "VID" of target signal from signaldata.csv
 - Search corresponding VID in vector.csv
 - vector.csv contains "Hang" and "Vang" value for each VID (This means each signal's hang and vang can be obtained from here)

NOTE: Autoware doesn't use vang value for traffic light recognition as vertical direction of signals is not important.

The definitions of value for Hang and Vang in vector.csv are:
 - Hang: North is 0 degree. Clockwise
 - Vang: Zenith is 0 degree. 

Best Regards

nsakar, 2016-08-08:
Thank you for your answer. I am presuming North is  x axis wrt car's orientation? Car's orientation = camera orientation. Since I am using ndt mapping based SLAM for map generation, I would presume x axis is north. Please confirm.
Also what is the precision level required to determine signal position? Can there be some x, y adjustment?

manato, 2016-08-08:
@nsakar ,
In Autoware, several coordinate systems are exist. Vector map coordinate are written in `/map` frame.
Camera coordinate system is not fixed on map coordinate system because camera coordinate system is defined by the position of camera, which moves on map.

Please find two pictures:

[image]

This shows the direction of our map coordinate system. This picture is north-up  and the correspondence between color and axis is:
 * Red = x
 * Green = y
 * Blue = z

So, North corresponds to y axis in our system.

[image]

On the other hand, this shows our camera coordinate system (color correspondence is the same). As you can see from this picture, z axis is the same as car direction.

> Can there be some x, y adjustment?
 
You can adjust x, y position of signals on image from [app] of `feat_proj`. `feat_proj` publishes the position of signals on image and this adjustment shifts them.

nsakar, 2016-08-14:
Will the camera coordinate system  be different from lidar coordinate system even if  camera is positioned front facing and close to the front facing lidar? Since the camera pose is computed  in the map frame of reference using published calibration file [ that establishes velodyne - camera transform], why would this change. My transform looks like follows:

 Translation: [-0.009, -0.024, 0.048]
- Rotation: in Quaternion [-0.514, 0.516, -0.477, -0.491]
            in RPY (radian) [2.935, -1.508, -1.380]
            in RPY (degree) [168.186, -86.403, -79.092]

closed 2016-08-20 by nsakar