autowarefoundation/autoware #424: ndt-mapping not working but lazy ndt-mapping works

Benjaminmar8, 2016-08-18:
Thanks for sharing the great work!

I am trying out the ndt-mapping using my own recorded bag file. My problem is, if I were to keep the tf between base_link to localizer as default to "1.2, 0, 2, 0, 0, 0" (this is done inside Autoware runtime manager ). The map doesn't build up and it only contains the first frame of the LIDAR scan. The transofrmation matrix is also constant. If I were to change the yaw, for example, to 1.57, the same first frame will be kept on adding to the output pcd map, with a translation in x or y direction. This problem exists for the moriyama bag file too.  However, lazy ndt-mapping works just fine. Could anyone help me on this issue, thanks a lot!

kitsukawa, 2016-08-19:
Can you show me the output on the terminal?

Benjaminmar8, 2016-08-22:
It's working now. It turns out that if I run the Autoware for the very first time, the parameters of the ndt-mapping are all set to 0, despite that these parameters has already been set to some default value in the run time manager. I only need to change the value a bit and press enter, the parameter values will then be loaded. 
Thanks a lot for the help.

closed 2016-08-26 by Benjaminmar8

sgzzgit, 2016-09-21:
Could you share the difference between ndt_mapping and lazy ndt_mapping. @Benjaminmar8 
Thanks.

kitsukawa, 2016-09-25:
Please refer to #453.