autowarefoundation/autoware #427: Transfering Pointcloud from sensor coordinate to world coordinate

news23, 2016-08-25:
Can Autoware transfer Pointcloud from sensor coordinate to GNSS coordinate?
I need to process Pointcloud data and GNSS data on same coordinate...
If it is possible, could tell me how to transfer the coordinate?

Thanks,

kitsukawa, 2016-08-25:
You can transfer the coordinate using TF package in ROS. In Autoware, we use TF to transform the coordinate.
Can you check the following link?
http://wiki.ros.org/tf

news23, 2016-09-05:
Thank you for the information

closed 2016-09-05 by kitsukawa

CowTC, 2018-03-02:
Hi, @kitsukawa.
Nice job, But i still have a question is that, when transform local coordinate to global coordinate,
We can use TF, but how to build the Rotation Matrix, And Translate Matrix.
3ks for your attention.