autowarefoundation/autoware #428: Split big map into smaller pcd files

Benjaminmar8, 2016-08-25:
I understand that the question has been discussed here #319 . However, the answer was not conclusive to me. I have a big bag file and it has been split into several smaller bag files. I can produce pcd files from the these smaller bag files. Based on the discussion of #319 , I will need to transform pcd files manually. 
Here is my question: how do I transform the pcd files. Assuming I have pcd1 and pcd2, what I am thinking is that I use ndt of PCL to determine the rigid transformation between pcd1 and pcd2, and then apply the transformation to pcd2 and save it as a new pcd (transformed pcd). Is this approach correct? Thanks a lot for the help.

kitsukawa, 2016-08-25:
Your approach is correct. I think that local2global node in ndt_localizer package may help you. The node does similar things you mentioned.

Benjaminmar8, 2016-08-26:
Thanks a lot for the answer. I am trying to figure out how to use the local2global node but still facing some issues. Assuming I have 5 pcd fils generated from 5 separate bag files, 001.pcd, 002.pcd, ... 005.pcd. First, I run 
          rosrun ndt_localizer local2global ~/002.pcd 0 0 0 0 0 0   
And then, I use Autoware to load the map 001.pcd. local2gloabl node will generate a "global_map.pcd", which is the transformed version of 002.pcd, and <x y z yaw pitch roll> parameters. 

After that, I run 
          rosrun ndt_localizer local2global ~/003.pcd <x y z yaw pitch roll>
and use Autoware to load map "global_map.pcd". The local2global node will generate a new "global_map.pcd" which is the transformed version of 003.pcd, and a new set of <x y z yaw pitch roll>. This continues until 005.pcd. Are these steps correct? Thanks a lot!

kitsukawa, 2016-08-26:
If you have five separate PCD files, you have to run local2global five times. You can modify the source code for convenience.

thomas3016, 2017-04-16:
@Benjaminmar8 I'm try to use your method above to stitching the pcd mapps, but  I can't get correct map, have you get a correct map?

yiakwy, 2018-11-26:
@Benjaminmar8 @kitsukawa  @thomas3016 is the problem resolved? I am going to submit a patch to resolve the problem, i.e. split a large PCD file (1 GB) to small files (tens of MB). Please let me know if you want my solution, or there is already a solution there.

kitsukawa, 2018-11-26:
@yiakwy I think you can solve the problem using PCD Grid Divider in Autoware.
https://github.com/CPFL/Autoware/tree/master/ros/src/util/packages/map_tools

yiakwy, 2018-11-26:
Roger. I will try to use it later. However, we are currently using PCL to process PCD data and I have little ideas about Autoware. 

I just scanned the codes and find that the following codes do the job.
```c++
    for (int i = 0; i < grid_num; i++) {
      if (grids[i].cloud.points.size() > 0) {
        pcl::io::savePCDFileBinary(grids[i].filename, grids[i].cloud);
        std::cout << "Wrote " << grids[i].cloud.points.size() << " points to "
                  << grids[i].filename << "." << std::endl;
        points_num += grids[i].cloud.points.size();
      }
    }
    std::cout << "Total points num: " << points_num << " points." << std::endl;
```

Could you tell me more about the relationship between PCL and Autoware?

kitsukawa, 2018-11-26:
Autoware uses PCL to process point cloud data. i.e. loading pcd, writing point cloud to PCD and so on.

yiakwy, 2018-11-27:
Please see my discussion:
https://github.com/PointCloudLibrary/pcl/issues/2652

closed 2019-03-01 by gbiggs