autowarefoundation/autoware #429: python module psutil.NUM_CPUS is deprecated

rookiepig, 2016-08-25:
In ubuntu 14.04, execute ***./run*** get the following error:

```bash
Traceback (most recent call last):
  File "/home/rookiepig/project/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 3157, in <module>
    app = MyApp(0)
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_core.py", line 7981, in __init__
    self._BootstrapApp()
  File "/usr/lib/python2.7/dist-packages/wx-2.8-gtk2-unicode/wx/_core.py", line 7555, in _BootstrapApp
    return _core_.PyApp__BootstrapApp(*args, **kwargs)
  File "/home/rookiepig/project/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2678, in OnInit
    frame_1 = MyFrame(None, wx.ID_ANY, "")
  File "/home/rookiepig/project/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 449, in __init__
    cpu_ibls = [ InfoBarLabel(self, 'CPU'+str(i)) for i in range(psutil.NUM_CPUS) ]
AttributeError: 'module' object has no attribute 'NUM_CPUS'
```

as suggested in this link [https://github.com/blaze/blaze/pull/1136], NUM_CPUS should be replaced with ***cpu_count()***

syohex, 2016-08-25:
I think you upgraded psutil version. Please downgrade psutil version. Ubuntu 14.04's python-psutil is 1.2.1. 1.2.1 does not implement 'cpu_count' method yet. If NUM_CPUS is replaced with `cpu_count()` like you said, it makes other users upgrade psutil version.

If Autoware updates required Ubuntu version like 16.04 or higher versions, then it is time to replace psutil method.

And I suppose you should not install package and upgrade system python packages. You should use [pyvenv](https://docs.python.org/3/library/venv.html) or [virtualenv](https://virtualenv.pypa.io/en/stable/)  in such cases

rookiepig, 2016-08-26:
I see. close this issue as solved.

closed 2016-08-26 by rookiepig