autowarefoundation/autoware #439: TF error in quick start

WeirdAI, 2016-09-06:
Thank you for your great work.
I've followed the quick start but there is a TF error in Points Raw. It says 

> For frame [velodyne]: No transform to fixed frame [world].  TF error: [Lookup would require extrapolation at time 1427158038.206998016, but only time 1427157650.349046016 is in the buffer, when looking up transform from frame [velodyne] to frame [world]]

So the quick start cannot work correctly. At the same time, there are TF errors in Vector Map, Vehicle Model and Global Waypoints.

> For frame [/map]: Fixed Frame [world] does not exist

> No transform from [base_link] to [world]

> For frame [map]: Fixed Frame [world] does not exist

I've uninstall the Velodyne PointCloud using `Sudo apt-get remove`.
By the way, I'm using Ubuntu 14.04 and ROS indigo. Waiting for your help. Thank you.

kshitijagrwl, 2017-01-31:
I'm getting the same error on following the quickstart guide. 
The system crashes while doing ndt_matching - 

```
-----------------------------------------------------------------
Log file: ndt_matching_20170131_052223.csv
use_gnss: 1
queue_size: 10
offset: linear
use_openmp: 0
get_height: 0
use_local_transform: 0
localizer: velodyne
(tf_x,tf_y,tf_z,tf_roll,tf_pitch,tf_yaw): (1.2, 0, 2, 0, 0, 0)
-----------------------------------------------------------------
['sh', '-c', '$(rospack find runtime_manager)/../../../.config/rviz/cmd.sh']
pid=14273
sched policy=OTHER prio=0
pid=14274
sched policy=OTHER prio=0
ndt_matching: /build/pcl-1.7-pY82A6/pcl-1.7-1.7.1/kdtree/include/pcl/kdtree/impl/kdtree_flann.hpp:172: 
int pcl::KdTreeFLANN<PointT, Dist>::radiusSearch(const PointT&, double, std::vector<int>&, std::vector<float>&, unsigned int) const [with PointT = pcl::PointXYZ; Dist = flann::L2_Simple<float>]: 
Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to radiusSearch!"' failed.
[ndt_matching-6] process has died [pid 13910, exit code -6, cmd /home/proto/Autoware/ros/devel/lib/ndt_localizer/ndt_matching __name:=ndt_matching __log:=/home/proto/.ros/log/17c25c80-e747-11e6-af16-525400db05a2/ndt_matching-6.log].
log file: /home/proto/.ros/log/17c25c80-e747-11e6-af16-525400db05a2/ndt_matching-6*.log

```

kitsukawa, 2017-02-10:
Can you compile Autoware again in the following way?
$ ./catkin_make_release
Sometimes, similar crashes occur after compiling using "$ catkin_make -DCMAKE_BUILD_TYPE=Release" .

closed 2017-05-29 by yk-fujii