autowarefoundation/autoware #440: An error which shows "cv has no member vector" happened when making release

lightlee14, 2016-09-07:
Hi, I meet an error with hint "error: 'vector' in namespace 'cv' does not name a template type" when I run command "./catkin_make_release". This error happened in "Building CXX object sensing/fusion/packages/calibration_camera_lidar/CMakeFiles/calibrationtoolkit.dir/CalibrationToolkit/calibrationtoolkit.cpp.o". After I modified all "cv::vector" to "std::vector" in calibrationtoolkit.h and calibrationtoolkit.cpp, it can been build. The vector should be "std::vector" or "cv::Vector"?

syohex, 2016-09-07:
Which OpenCV version did you use ?

> OpenCV 2.4.10 or higher NOTE: Autoware does not support OpenCV 3. Please use OpenCV 2

sujiwo, 2016-09-07:
I think this is related to compiler version; do you use GCC 5.x? I got this
error when using Ubuntu 16.04, which does not exists in 14.04 (GCC 4.8).

syohex, 2016-09-07:
No. However Autoware does not support Ubuntu 16.04 yet.

lightlee14, 2016-09-07:
Thanks a lot. I use Ubuntu 16.04 && g++ 5.4.0 && OpenCV 3.1.0 of ROS kinetic. Also I built OpenCV 2.4.13 alone and modified CMakeLists.txt to use OpenCV 2.4.13 to build whole project. While the CUDA of the hardware only support Ubuntu 16.04, does it exist any way to build Autoware under Ubuntu 16.04?

syohex, 2016-09-07:
I'm not sure there is no way now. However Autoware supports it someday.

lightlee14, 2016-09-07:
Thanks a lot!

closed 2016-11-21 by amc-nu