autowarefoundation/autoware #444: No Points Raw in Quick Start. Somebody help me. For frame [velodyne]: No transform to fixed frame [world].

WeirdAI, 2016-09-14:
`For frame [velodyne]: No transform to fixed frame [world].  TF error: [Lookup would require extrapolation at time 1427157777.666513920, but only time 1427157654.152538880 is in the buffer, when looking up transform from frame [velodyne] to frame [world]]`

sgzzgit, 2016-09-21:
TF

WeirdAI, 2016-09-21:
Yeah. I know it's TF error. But I do follow the Quick Start guide exactly. How can this error happen and how can I fix it. Can you help?

kitsukawa, 2016-09-25:
Please try to click "Reset" button on RViz. Sometimes, TF error is fixed by clicking "Reset" button.

WeirdAI, 2016-10-13:
Thank you for your help, but it does not work. The problem is still there. When I use roswtf, it says the ndet matching died. What's radiculous, I do the same processures on two different laptops; the former one works well, but the latter one comeup with this problem.

kfunaoka, 2018-07-13:
Still not working? Please check https://github.com/CPFL/Autoware/wiki/Demo and try again.

closed 2018-07-13 by kfunaoka