autowarefoundation/autoware #453: Difference between ndt_mapping and lazy ndt_mapping

sgzzgit, 2016-09-21:
Could anyone introduce the difference between ndt_mapping and lazy ndt_mapping when using it with velodyne lidar. Thanks very much.

Benjaminmar8, 2016-09-22:
I might be wrong, but my guess is lazy ndt_mapping does not use all point cloud frames, while ndt_mapping uses all point cloud frames. 

I did try both of them on a large data set collected using velodyne 32E, lazy ndt_mapping some times has large drift in z axis and the amount of drift is not consistent.

sgzzgit, 2016-09-24:
Thanks for your reply. @Benjaminmar8 
My lidar is Velodyne HDL-32 too.Maybe you and I are facing the same problem below.
When mapping in a larger place ,like a garden. ndt_mapping must run a very long time. At least two days. It is very troublesome to me. Have you ever met this problem?

kitsukawa, 2016-09-25:
As @Benjaminmar8 guessed, lazy_ndt_mapping does not use all points for matching while ndt_mapping uses all the points as reference. lazy_ndt_mapping uses only three previous scans as reference by default. lazy_ndt_mapping is faster than ndt_mapping, but the output is low accuracy.

Benjaminmar8, 2016-09-26:
We might have a similar setup. @sgzzgit 
My issue is that I am unable to build a large-scale, say 1km by 1km map, though smaller maps seems fine. My approach is to build smaller maps and stich them together. I had limited success, but towards the end I noticed that the last map and the first map didnot overlap (thus loop closure). The error is large, looks like ore than 10 meters in translation and more than 40 degress in rotation. I wasn't sure if this is due to the error in ndt-mapping or it is sth. that I did wrongly. 
How do you manage a big map?

sgzzgit, 2016-09-26:
> My approach is to build smaller maps and stich them together

I am very curious how you joint the small maps together?  @Benjaminmar8  I had tried the same idea before, but failed to joint them together.
Up to now, the biggest map I had built is 360m * 300m. It could be used to localize very well.
However I want to build a map whose size is 800m * 790m now.
After capturing HDL-32 lidar data, I launch ndt_mapping, unfortunately ndt_mapping and lazy ndt_mapping  crashed everytime. I am very confused. Maybe we can discuss how to joint small maps together like the demo data.

sgzzgit, 2016-09-27:
Have you ever analyzed the .pcd file of pointcloud_map in the sample_moriyama_data? @Benjaminmar8

Benjaminmar8, 2016-09-27:
@sgzzgit 
I didnot analyze that map file as I do not know the ground truth.

sgzzgit, 2016-09-27:
I analyzed the map file,and now i plan to joint small map files.
Do you have any other suggestions? @Benjaminmar8 
Thanks very much.

Benjaminmar8, 2016-09-27:
you may refer to this question #428

Benjaminmar8, 2016-09-27:
@sgzzgit 
By the way, how do you analyze the map file?

closed 2016-10-07 by sgzzgit