autowarefoundation/autoware #457: Is NDT-mapping able to do loop clousre?

Benjaminmar8, 2016-09-23:
Dear All, I am trying to build a large map by using Velodyne 32E mounted on a car. The start point and end point are at the same place. I have recorded the bag file and used NDT-mapping to build the map. However, I noticed on the map that the loop was not closed properly. My map is about 1Km by 1Km. My question is, does ndt-mapping support loop-closure Thanks a lot!

sgzzgit, 2016-09-24:
Yes. It could

shinpei0208, 2016-09-24:
No, it does not support loop closing now. We are adding that capability.
Thus, the map of 1km long must include errors. We suggest users to get a
map from the map company, if a large area is needed.

sgzzgit, 2016-09-24:
> Thus, the map of 1km long must include errors. 
Sorry to bother you.   @shinpei0208 
You mean the distance must smaller than 1Km when using ndt_mapping? If I improve the speed of the vehicle when capture lidar data, will the distance be larger than 1 Km?

Hoping for your reply.

shinpei0208, 2016-09-24:
ndt_mapping, as it is, does not support loop closing. Therefore, it
accumulates errors while mapping. Assume for example that an error of 0.1cm
is accumulated every 1m distance. If the distance of map is short, errors
are small. But if it goes over a few kilometers, errors cannot be ignored.
Under the above assumption, 1km distance of map could include 1m error.
Loop closing would solve this problem.

kitsukawa, 2018-06-05:
@anhnv-3991 Can you answer the question above?

anhnv3991, 2018-06-05:
@witchoco 
There are two versions of ndt: running on the cpu and running on the gpu. The original ndt is quite slow, so we tried to speed it up by using gpu. Since we couldn't find any gpu-version ndt from the pcl library (it is said that they were work-in-progress), we make our own version (including gpu octree, registration, and voxelgrid).
To compare the performance of the cpu and gpu, algorithms should be similar between them. Since the pcl use kdtree for the nearest neighbor search in the ndt, compare the original ndt with gpu seems unfair. Hence we decided to make a nother cpu version of ndt which uses octree for the search.
As for the comparison with PCL, yes, as our evaluation indicated, our implementation is 2x faster for the cpu, and 3.5 - 4x faster for the gpu.

witchoco, 2018-06-05:
@kitsukawa  @anhnv3991 
Got it. Thank you so much. :)

kfunaoka, 2018-07-13:
Registered as enhancement at #1349

closed 2018-07-13 by kfunaoka